Ultimaker/Cura #10152: Filament profile should override the settings profile.

robinstoraa, 2021-07-18:
### Is your feature request related to a problem?

Yes. There is a problem. Its hard to use cura with alot of different filaments that need different flow rates, temperatures, retractions, speeds etc.

### Describe the solution you'd like

I would like the material profiles to have a section where you can set any override over the printing profile. As of now you can only override a couple of settings, but not all of them. Please make it so that the material profile can override any setting in the printing profile. So that for example you an have a specific speed setting for each filament or a flow setting for each filament, or a different cooling amount for each filament etc. 

### Describe alternatives you've considered

I haven't thought of any alternatives, as my main solution should be easy to integrate. Replace contents of the material "Print Settings" menu with a (+) button where you could select what setting to override instead of only showing the couple that it is showing as of now.

### Affected users and/or printers

Everyone will benefit from this solution. It will be easier for anyone using alot of different filaments, especially different filament types, like flexibles that need more settings adjusted than regular filaments.

### Additional information & file uploads

[image]
Remove the settings in the menu and add a (+) button. The button should pop up another menu with a list of all the settings you have set to show. Then select what settings you wish to override.

nallath, 2021-07-19:
Without a complete re-write of the current architecture, it's impossible for material profiles to fully override settings changed in the quality profiles (see https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings for more info).

You can install a plugin that allows for all settings to be added to a material profile: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MaterialSettingsPlugin This is a plugin made by a comunity member, so if you like his work (and can spare it) please consider leaving him a donation.

Ghostkeeper, 2021-07-26:
The title of this issue seems to indicate something completely different from its contents.

Ghostkeeper, 2021-07-26:
I'll defer this issue since the Material Settings plug-in already implements this feature, albeit slightly different on the details.

It is an issue that we could address in the future, but for now we won't give it priority over things already in our backlog.

closed 2023-07-13 by github-actions[bot]