Ultimaker/Cura #10153: Avoid Printer Part when travel

cvignali, 2021-07-18:
### Application Version

4.9.10

### Platform

Windows 10

### Printer

Ender 5 pro

### Reproduction steps

In general should be easy to reproduce because this is something that has been bothering me long time. 
= Download https://www.thingiverse.com/thing:4727448
- Sclae 150%
- Slice it, and check travels around 50%.
- Ensure Avoid Printer Part when travel is ON.

### Actual results

Even when "Avoid Printer Part when travel" is on, the travel is traversing the recently printed layer causing an annoying noise that not until today I guess it make the Step motor to fail and lose positioning. 
You will see that the is a layer that cause a huge imperfection. 
In many occasions this problem cause that printing part get loosen because the Hot end hits some part and break it lose. 
It is an annoying but causing many issues to me, and I am sure to many users. 
You will see in the video that noise when the hotend goes through the printer parts. 
https://photos.app.goo.gl/vWwZE47pQwyMNavL6

### Expected results

If that option is selected, the printer should avoid recently printer parts that have no tolerance to move the hotend over it. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

cvignali, 2021-07-18:
[image]

fvrmr, 2021-07-19:
Hi @cvignali the option avoid printed parts when traveling is more about other parts that would be on the buildplate. Read here more about this setting: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/travel/travel_avoid_other_parts.md

But you could use combing mode: within infill. If you don't want that the travel moves go over the model. Let me know if that works for you!

Ghostkeeper, 2021-07-26:
Well, in this case, from your screenshot, it looks like there is simply no path possible that avoids the previously printed parts.

You could try reducing the Travel Avoid Distance setting so that it is allowed to get closer to the model, and hopefully travel between the ribs of that slug. But otherwise there simply is no possible path to get from where it ended up to where it needs to be. If there is no path that doesn't cross it, it'll simply have to cross it.

Alternatively, enabling Z hop could also solve your issue.

As far as I can see, this is expected behaviour.

closed 2021-07-26 by Ghostkeeper