Ultimaker/Cura #10154: Questions about double spray

jwh66, 2021-07-19:
### Is your feature request related to a problem?

I encountered a problem when printing with non-independent dual nozzles. When the No. 1 nozzle was working, the No. 2 nozzle had excess material oozing out. I tried to modify the pump back, but there was always a oozing problem. So on the basis of filling the tower, I additionally opened the seepage cover, but I found that cura did not make effective use of the tower and seepage cover. For example, the current one hundredth floor, No. 1 nozzle will give priority to complete the tower and exudation hood before printing the model, which can effectively offset the excess material exuded by No. 2 nozzle, but only the tower is printed when the No. 2 nozzle is switched. The exudation cover is not used, and the excess material exuded from the No. 1 nozzle will be stuck on my model.

### Describe the solution you'd like

For example, for the 100th layer, when the No. 1 nozzle prints the tower and the exudation cover, two layers are printed: the 100th layer and the 101st layer model, so that when the No. 2 nozzle is switched, the 2nd layer can be printed more. One time oozing out of the hood, because the height of the model is one hundred and one layer

### Describe alternatives you've considered

Change the shape of the loading tower, such as square

### Affected users and/or printers

Everyone who uses dual nozzles
At least we don’t need to spend a lot of time repairing my model

fvrmr, 2021-07-19:
Hi @jwh66 thank you for your feature request. 
So you would like to define yourself when extruder 1 or 2 are starting to print? Like this feature request: https://github.com/Ultimaker/Cura/issues/9960 
Let me know if that is what you would like to have or not.

jwh66, 2021-07-20:
> 嗨@jwh66，感谢您的功能请求。
> 那么您想这个在机上 1 或 2 开始打印时定义自己吗？喜欢功能请求：#9960
> 让我知道这是否是您想要的。

jwh66, 2021-07-20:
> 嗨@jwh66感谢您的功能请求。
> 那么您想在挤出机 1 或 2 开始打印时定义自己吗？喜欢这个功能请求：#9960
> 让我知道这是否是您想要的。

I think that when I use the non-independent dual nozzles, it can reduce the seepage of the inactive nozzles, because the inactive nozzles often hang the material on the surface of my model. I tried to open the seepage cover and the tower together, but I found cura The calculations did not effectively use the seepage hood and tower.
ps Can I try to add dwell time when switching nozzles? A custom one

closed 2021-07-20 by jwh66

Ghostkeeper, 2021-07-26:
> but only the tower is printed when the No. 2 nozzle is switched. The exudation cover is not used, and the excess material exuded from the No. 1 nozzle will be stuck on my model.

The Ooze Shield is only printed once per layer. Two ooze shields would have the same height as one ooze shield, so they would not scrape any more ooze from the nozzle than one ooze shield would.

Maybe you're trying to use the ooze shield to prime. It is not made for that. It is only made to protect against ooze. If the highest layer of the ooze shield was printed with the other extruder, it would still protect against ooze the same way.