Ultimaker/Cura #10156: printing fail

TDW-123, 2021-07-19:
### Application Version

4.10.0

### Platform

windows 10

### Printer

Anet ET4 Pro

### Reproduction steps

ik heb een normale file in cura gezet dan wat verkleint.

### Actual results

De file begon te printen van 30% en dat blijft gebeuren met meerdere files dat ik probeer.

### Expected results

Hebben jullie misschien een verklaaring/oplosing oor mijn probleem?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Black+Mamba+Spinning+Top.zip](https://github.com/Ultimaker/Cura/files/6843269/Black%2BMamba%2BSpinning%2BTop.zip)

fvrmr, 2021-07-20:
Hi @TDW-123 wat bedoel je precies met begon te printen van 30%? Zie je dit ook terug in de print preview? Foto's of screenshots zijn hier dan extra handig. 
En de bestanden die je hebt gedeelt zijn geen project files, dus kan niet goed reproduceren. Een project file bevat namelijk alle settings en printer die je gebruikt hebt. Een project opslaan gaat via File - save project.

TDW-123, 2021-07-22:
Dit is aan het gebeuren de hele tijd
https://user-images.githubusercontent.com/62179187/126659501-7dfc99d9-cda4-4e3f-87b2-5510b683f806.mp4

Ghostkeeper, 2021-07-26:
Please keep bug reports in English. Many Cura developers can speak Dutch due to Ultimaker being located in the Netherlands, but if it's English the rest of the world can also follow along.

Neither of your uploaded files contain any project files by which we could reproduce the issue if the g-code was wrong.

However you did provide the resulting g-code. Your video seems to show (if I interpreted it correctly) that your printer simply skips over the first 70% of the file. However there is no instruction in the g-code to do that.

There's two things here that I can't know yet:
* Is the printer actually skipping a part of the g-code and starting to print in mid-air, or is it just the progress bar being wrong?
* In the latter case, I don't know if the progress bar is being generated based on the layer or if the printer firmware actually uses the time estimates from the g-code comments that Cura puts in there.

The file `AET4PRO_TDW v1.gcode` contains 7 layers. Each of them is labelled with `;LAYER:#` so everything seems to be working fine on Cura's side there. The g-code is also annotated with `;TIME:1258` at the start and `;TIME_ELAPSED:###.######` at every layer. So that seems to be fine as well.

If the printer is actually skipping part of the g-code, I suggest that you try using a different SD card or USB drive to transfer the g-code to the printer, and that you properly eject the drive before physically remove it from the computer. Not doing so can result in incomplete files.

If it's only the progress bar, I just have to blame the firmware. There doesn't seem to be anything wrong with the g-code you included in the zip archives. Or can you indicate what is wrong with Cura's output?

closed 2024-12-09 by github-actions[bot]