Ultimaker/Cura #10157: Cannot Save Gcode if Support Extruder is 1

dotdash32, 2021-07-19:
### Application Version

4.10.0

### Platform

MacOS

### Printer

Custom FFF

### Reproduction steps

1. Load model onto printer with 2 extruders
2. Select model to be on Extruder 2
3. Enable Support Generation, use Exutrder 1
4. Slice
5. Send Save to Removable Drive/File

### Actual results

Popup with "Error: 0" and loading bar saying "saving."  Blue indicator bounces around for a bit, but then popups disappear, and file is empty in removable driver, does not appear at all when "saving to disk."

However, reverting Support Extruder to 2 (my profile default) or switching the main model to Ex 1 allows saving to proceed normally.  Also tried to add another model in Ex 1, but problem still exists.

Low priority, as I can get around it, but seems weird I can't use flip the extruders around.

### Expected results

File saves Gcode normally.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6843599/cura.log)

I can't save the project file with the model in Ex 2, so will need to be changed afterward (SupportTest, select Extruder 2).

[saveIssue.3mf.zip](https://github.com/Ultimaker/Cura/files/6843609/saveIssue.3mf.zip)

dotdash32, 2021-07-19:
Some additional testing after a shower thought: 

I was able to save a print gcode yesterday when I had Build Plate Adhesion Extruder set to 2 and Support Extruder set to 2.  Today I did some testing, and when build plate adhesion extruder = model extruder, it seems to save without issue.  When BPA extruder is a different extruder than the model, it only works the support extruder is the same as the model extruder.  

The table below summarizes the results of my testing.

 |Model Extruder | Support Extruder | Build Plate Adhesion | Can Save?|
|----|---|---|---|
| 1 | 1 | 1 | Yes |
| 1 | 1 | 2 | Yes |
| 1 | 2 | 1 | Yes |
| 1 | 2 | 2 | NO |
| 2 | 2 | 2 | Yes |
| 2 | 2 | 1 | Yes |
| 2 | 1 | 1 | NO |
| 2 | 1 | 2 | Yes |

It seems that when the Model Extruder =/= (Support Extruder == Build Plate Adhesion Extruder), there is a problem.  I am unsure if this extends to changing infill extruder settings or not, have not tested overriding extruders for other elements.

In the interest of more testing, it works with a Ultimaker 3/3 Extended profile in the same 1 | 2 | 2 configuration, so might be either the Custom FFF environemnt, or something else in my profile.  Please see below for working Ultimaker project file.

[testSave.3mf.zip](https://github.com/Ultimaker/Cura/files/6844447/testSave.3mf.zip)

fvrmr, 2021-07-20:
Hi @dotdash32 thank you for your detailed bug report. Unfortunately I can't reproduce your issue on Windows or macOS. What version of MacOS are you using? And does it also happens when you want to save it to disk or only a removable drive?

dotdash32, 2021-07-20:
I'm on macOS Big Sur (11.4) on a 15" MBP from Mid 2015.  It happened for both save to disk and removable drive, but it was faster to test with a removable drive since I didn't need to choose a folder to save to each time.  It also affected my ability to save the project (as a 3mf), but that that was not tested as thoroughly.

fvrmr, 2021-07-21:
Could you try to disable the Linear Advance Setting Plugin? 

This is what I found in the log:

```
2021-07-19 13:00:55,477 - ERROR - [MainThread] UM.Logger.logException [112]: Exception: Unable to write to file support test: 0
2021-07-19 13:00:55,480 - ERROR - [MainThread] UM.Logger.logException [116]: Traceback (most recent call last):
2021-07-19 13:00:55,482 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/UM/Qt/Bindings/OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2021-07-19 13:00:55,484 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDriveOutputDevice.py", line 99, in requestWrite
2021-07-19 13:00:55,487 - ERROR - [MainThread] UM.Logger.logException [116]:     self.writeStarted.emit(self)
2021-07-19 13:00:55,489 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/UM/Signal.py", line 219, in emit
2021-07-19 13:00:55,491 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/UM/Signal.py", line 336, in __performEmit
2021-07-19 13:00:55,493 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/UM/Signal.py", line 219, in emit
2021-07-19 13:00:55,496 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.10/build/inst/lib/python3.8/site-packages/UM/Signal.py", line 332, in __performEmit
2021-07-19 13:00:55,498 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/<user>/Library/Application Support/cura/4.10/plugins/LinearAdvanceSettingPlugin/LinearAdvanceSettingPlugin/LinearAdvanceSettingPlugin.py", line 191, in _filterGcode
2021-07-19 13:00:55,500 - ERROR - [MainThread] UM.Logger.logException [116]:     if not apply_factor_per_feature[extruder_nr]:
2021-07-19 13:00:55,503 - ERROR - [MainThread] UM.Logger.logException [116]: KeyError: 0
2021-07-19 13:00:55,955 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [289]: SliceInfo sent successfully
2021-07-19 13:00:57,568 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///Applications/Ultimaker Cura.app/Contents/Resources/qml/UM/MessageStack.qml:75: TypeError: Cannot read property 'horizontalCenter' of null
```

dotdash32, 2021-07-21:
Yes!  Disabling the Linear Advance plugin does fix this. It appears this has also been fixed in [LinAdv's #31](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/issues/31), and is just waiting for a release in the Cura marketplace.  I just tested the added line, and it does work.  So the solution is just waiting for release.

When I disabled and then re-enabled, the error switched from 0 to 1 (seems minor).

I was able to get it to save (without plugin modification) by changing the Linear Advance per type manually, which seems to avoid the `apply_factor_per_feature` issue.  However, the easiest way to fix this without waiting is to just add the line

```python
apply_factor_per_feature[extruder_nr] = False
```

at line 159 in the Plugin python script.

Ghostkeeper, 2021-07-26:
This is not an issue in Cura then but in a plug-in (and luckily already fixed there). Thanks for the investigation!

closed 2021-07-26 by Ghostkeeper