Ultimaker/Cura #10158: Strange shapes appear in sliced model on certain layer heights

KenjiTakahashi, 2021-07-19:
### Application Version

4.10.0

### Platform

Archlinux

### Printer

Ender 3

### Reproduction steps

Just slicing as usual :-). And getting very strange triangular shapes appearing at certain layer height settings. See screenshots below.

### Actual results

Project 1 sliced result:
[image]
Project 2 sliced result:
[image]

### Expected results

The sliced models to be like the actual models, without artifacts inside.
Project 1 model:
[image]
Project 2 model:
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

First project files: [wth.zip](https://github.com/Ultimaker/Cura/files/6844413/wth.zip) (fails at `0.2`, works at other heights like `.0.16`, `0.24`, etc.).
Second project files: [wth2.zip](https://github.com/Ultimaker/Cura/files/6844433/wth2.zip) (fails at `0.16` and `0.2`, seems to work at higher values).

nallath, 2021-07-20:
I think this might be a duplicate of https://github.com/Ultimaker/Cura/issues/9979

fvrmr, 2021-07-20:
I will close this issue since it is indeed a duplicate as mentioned above. Please continue the thread there, so we keep everything in one place.

closed 2021-07-20 by fvrmr