Ultimaker/Cura #10160: Partial missing layer after slicing...

marcdraco, 2021-07-20:
### Application Version

4.10.0

### Platform

Linux

### Printer

Ender 5

### Reproduction steps

Sliced with tree supports

### Actual results

Sliced normally until I was several hours into the build an then I was horrified to notice there are missing layers! I can probably fill them so I'll have to let it run and keep my fingers crossed.

### Expected results

Expected a complete shell
[image]
[cura.zip](https://github.com/Ultimaker/Cura/files/6850161/cura.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I've see odd missing layers since I started using 4.10 but I'd assumed it had been problems with my printer jamming (which it has been) - I've resolved those issues and just happened to spot this on my screen by chance!

fvrmr, 2021-07-21:
Hi @marcdraco I can't reproduce your issue. It could be that it is related to https://github.com/Ultimaker/Cura/issues/7412 can you check if the missing layer is printer at layer 0?

marcdraco, 2021-07-21:
I'm sorry to sound dense but I'm unsure what you mean by this issue. I'll follow with some pictures of the actual model as printed which have the the missing layers exactly as the project preview predicts.

This is my third attempt at printing this part (which at about 240g so not exactly cost-free but "dem's da breaks"!) and I was so convinced the problem was with the hardware that I replaced the bowden with brand new Capricorn, toolhead from stock V8 to an E3D V6, tightened the belts like garottes (perhaps that language is a bit colourful, but you get the idea), tuned extrusions to a fraction of a percent, turned the speed back down --- pretty much you name it. Result is always the same. 

Assuming this current run completes I'll have to use epoxy to prevent the model from coming apart in use - this has only started with V4.10 so far as I know, I'll d/l 4.8 or 4.9 later to see if I can reproduce this.

There's an extra "layer" of what looks like support in there too I think, I've not really paid it that much heed since I was concentrating on the hardware issues. 

I've learned a lot more about the Ender 5 and Klipper as a direct result, so it's not all bad! :)

marcdraco, 2021-07-21:
Here are a couple more snaps from the sliced model - actual physical pictures to follow. These holes should be circular!

[image]

[image]

marcdraco, 2021-07-21:
[image]

Seems I was mistaken - this issue is present in 4.9 too! I've checked in the latest 4.10-beta and it's still there. This view shows the inside of the model and you can see there are strange diagonals across where those missing layers are appearing.

marcdraco, 2021-07-21:
[image]
[image]

Oh.. and 4.8 - which leads me to believe there is a peculiarity in the .STL file which isn't showing in the preview window.

And here in Blender 2.9 just to see if there's anything obvious spooky going on in the STL.

[image]

Asterchades, 2021-07-23:
3D Builder finds invalid geometry in the STL, and MeshMixer basically had a coronary when I tried to find out where exactly the problems are.

In the interest of simplicity I'll upload the results I got from 3D Builder for you to try. If they work as expected, this is simply a case of a faulty model. That it looks fine in Blender is more likely to do with its origins not being associated with creating objects that have to exist in the physical world.
[missing-layer (fixed).zip](https://github.com/Ultimaker/Cura/files/6866791/missing-layer.fixed.zip)

marcdraco, 2021-07-23:
Thank you for that, but this wasn't designed in Blender - I was simply using Blender as a different example of an STL import. 

This model (which is part of a larger design) was created in FreeCAD 1.8 and latterly 1.9. (The reasons you mentioned are why I moved from a less "physible" system like Blender which I'm more comfortable with to a hard-edged CAD system which "thinks" in terms of a complete object define mathematically, vs. a series of polys manipulated over time.)

Prusa Slicer (2.3.2) also renders it without that error. 
MeshLab (v2020.12) similarly has no problems. 
Perhaps unsurprisingly, FreeCAD's STL importer also imports and renders it without issue.

This makes me question the origins of the various STL import/export engines - if they are all derived from an original piece of FOSS software (or perhaps two different ones) that might explain it... if not? This is anyone's guess and we're no further forward.

Asterchades, 2021-07-24:
Unfortunately I'm with @fvrmr on this one. I loaded the original model into 4.10.0, using the stock "Custom FFF/FDM Printer" profile on Fine (ie Everything left at absolute Cura defaults, per Ultimaker's specs), and am completely unable to reproduce the issue even using various combinations of Mesh Fixes. _Though it is worth mentioning that 4.10.0 is only importing the model, not the whole project, and I'm unable to force it to do otherwise._

The same is true importing it into SmartAvionics' custom builds, which is letting me import all the printer and filament settings at the same time. Again messing around with various Mesh Fixes yielded no unexpected changes.

marcdraco, 2021-07-24:
This is most perturbing but that's computers for ya! :)

The project an "improved" Ghostbusters 1 Proton Pack "wand" (I don't know the lingo, I'm just the idiot tasked with creating it from basic plans ;) ) is completely Open Source hardware and physibles, it's just not sufficiently complete that I can release it as yet. There's no point in giving everyone a partly finished, untested model because it devours filament. The current backpack design (not mine) comes in at around several kilos and that's with the support material removed. 

The design lacked some features that my little "mini me's" significant other wanted for his cosplay and as a result it's necessitated a complete re-working of the original in a clean room so we don't tread on toes and although there are similar designs, I think this one improves on them all in affordable and easily constructed ways.

Fortunately the backpack has been relatively easy to adapt with some extra goodies here and there so at least I don't have to do that from the ground up!

Working alone in my spare time with electronics knowledge that's over 40 years out of date (dang, I'm old!) has slowed progress somewhat but I'm slowly catching up with these new fangled MOSFET things. (I joke of course, but in my day, IGFETs as we called them then, were only just starting to enter the mainstream and germanium was still king. I only recently discovered IGBJTs too. Snooze and you lose in this game!) 

As you might imagine being able to squeeze what amounted to a fairly complex and detailed piece of electronics and hardware when Ghostbusters was made into something about the size of box of matches (but without resorting to SMT) blows my mind. 

TL;DR

Thank you for your assist here - we'll just have to run with it for now and see if anyone else reports problems. It's entirely possible that using different slicer settings this will work just fine or when I make some small changes to dimensions here to allow for printing tolerance that this will just vanish from whence it came. Begone foul bug...

closed 2021-07-24 by marcdraco

Ghostkeeper, 2021-07-27:
I think we might've got another duplicate of https://github.com/Ultimaker/Cura/issues/9979 here. It's been seen a couple of times, but only with Linux.

marcdraco, 2021-08-31:
Looks like it @Ghostkeeper - there's a glitch in the Linux matrix but knowing this ahead of time certainly makes it easier to work around!