Ultimaker/Cura #10161: two printhead print cant save

rupsimrd, 2021-07-21:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

custom

### Reproduction steps

1. imported models, selected which head prints what, merged models
2. sliced model without problems
3. tried to save to computer, got this error
[image]
4. tried to save project, got the same error
5. tried to save to sd card, got empty 0B gcode file

### Actual results

error 
0

### Expected results

saved gcode

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log.1.zip](https://github.com/Ultimaker/Cura/files/6853473/cura.log.1.zip)

nallath, 2021-07-21:
The only error that I see in the log is about one of the wall line width settings being zero.

rupsimrd, 2021-07-21:
> width

all line widths are set correctly for both toolheads

nallath, 2021-07-21:
Okay, could you share a project that reproduces this?

fvrmr, 2021-07-21:
Seems like this is a duplicate of https://github.com/Ultimaker/Cura/issues/10157 and is probably related to the Linear Advance Setting plugin. Could you try to disable the plugin and see if that resolves your issue. 
I will close this issue, but please continue the thread in the other issue.

closed 2021-07-21 by fvrmr

rupsimrd, 2021-07-21:
yup that was it thanks!