Ultimaker/Cura #10165: Cura AppImage not working on Latest fully Updated Ubuntu 20.04 LTS

yash-fn, 2021-07-21:
### Application Version

4.10.0

### Platform

Ubuntu LTS 20.04

### Printer

Irrelevant

### Reproduction steps

Download latest AppImage from Cura Website
chmod +x on AppImage
./Ultimaker_Cura-4.10.0.AppImage

### Actual results

./Ultimaker_Cura-4.10.0.AppImage show the following on the command line:
```
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
```

and the following GUI popup:

```
Failed to probe OpenGL

Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.
```

Downgrading to the 4.9 version works.

### Expected results

Cura should open and run. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

yash-fn, 2021-07-21:
Okay. This appears to be a system bug and not a cura bug. Closing. Somehow it works on another ubuntu computer I have but just not my desktop computer.

closed 2021-07-21 by yash-fn

Ghostkeeper, 2021-07-28:
You'll get this error if Cura is unable to get the graphics driver to create an OpenGL context with OpenGL2. So either you'd have an extremely old OpenGL version (unlikely 17 years after its release) or it's just not able to get the graphics driver.

I've seen this myself sometimes if I just updated my NVidia graphics drivers and the updater asks me to restart my computer, but I didn't restart my computer yet. If it's consistent for you, I think you may just have a broken graphics driver or no driver at all. Maybe you're running through a VM and that VM doesn't support hardware acceleration?

yash-fn, 2021-07-28:
Ah interesting. So I purged my nvidia drivers then reinstalled it and then restarted computer and now it works. Looks like you were right. The nvidia driver may have not been properly setup or something. Either way thanks!