Ultimaker/Cura #10166: Brim clipped to build area

nathanielstenzel, 2021-07-21:
### Is your feature request related to a problem?

When you add on brim to a print, you decrease the size of the stuff you can print. 

### Describe the solution you'd like

Could you clip the brim at the edge of the build area? That way we could go ahead and print stuff at full size with as much brim as we figure we would need and we would only have the problem that areas where the part is too close to the edge of the build area might not stick as well. You could perhaps increase the brim size of some of the rest of the print in the general area to make up for the lack of brim near the edge.

### Describe alternatives you've considered

You could have an option to restrict the brim to areas between parts of the actual print so that if you wrap a line around the print the brim would be somewhere inside the line. While this might not be as effective, a person could choose to have all of such areas be a big brim which at that point is basically a 1 layer raft. If it is all of the area or just a number of mm/lines should probably be up to them.

### Affected users and/or printers

all printers

fvrmr, 2021-07-22:
Hi @nathanielstenzel thank you for your feature request.
I will bring this up for discussion with the team. Keep you posted!

nallath, 2021-07-23:
I've created a ticket on our backlog: CURA-8433

nathanielstenzel, 2021-07-23:
Thank you much. This has been an obstacle for me since I started 3D printing back in 2014. It will he nice to not have sacrifice my precious brims anymore.

nallath, 2021-07-26:
Just a bit of expectation management: Just because it made it to the backlog doesn't mean that we will do it any time soon (or at all) though. It just means that we felt that it has enough merit to "keep it in the race".

Ghostkeeper, 2021-07-28:
This feature is suffering from the same difficulty as https://github.com/Ultimaker/Cura/issues/4222, where we'd have to be able to print brims as polylines in order to implement this.

nathanielstenzel, 2021-07-28:
Ah. So brim control is a problem for many issues in Cuba. Just be be sure I understand everything clearly, if I replace the word "brim" with "skirt" then it would still not be able to work like I want or be able to adjust the code to do so, right?

Ghostkeeper, 2021-07-29:
No indeed. The brim and the skirt are generated with exactly the same algorithm in CuraEngine. The only real difference is which settings they listen to and that the skirt gets a morphological close operation of 10cm, which cuts off inner corners for the most part.

Ghostkeeper, 2021-09-07:
> Maybe having a brim generated as a solid single layer raft with a smoothing of 10mm and some dots added in to extend the rafty brim would do the trick for what I was looking for.

The raft also has a morphological close similar to the skirt, but with configurable radius (the skirt uses 100mm). That's what the raft smoothing does. Something similar is requested in this feature request: https://github.com/Ultimaker/Cura/issues/7439

However that has little to do with clipping the brim to the build volume, which was the original feature request.

nathanielstenzel, 2021-09-07:
I may have been unclear. It is not like I literally want a raft. It is just that the shape of the raft that I can get when adding some simple dots to the build plate is about right else than the fact that it goes under the object too.

nathanielstenzel, 2021-09-16:
@Ghostkeeper well, I was told the requested adjustment was not possible and I considered a possible alternative, as I mentioned in the post 20 days ago. I always look for alternative ways to get what I want when I run into a brick wall. Looking at things from different angles or redefining the problem is just something that I tend to do. I am a programmer too, but I doubt I could locate and understand the code for the brim algorithm without a while of poking around at it. I might step on someone's toes if I was to do that too.

nathanielstenzel, 2021-09-16:
https://github.com/Ultimaker/Cura/blob/master/cura/BuildVolume.py makes some references but I guess https://github.com/Ultimaker/CuraEngine/blob/master/src/SkirtBrim.cpp is really where I should look if I do want to meddle with your code. Which means I just cloned the wrong project. hahaha Figures. I should have known the Cura slicer app would also just be a pretty interface to using the Cura Engine. That is probably as it should be anyways.

nathanielstenzel, 2021-09-16:
Ah. I see. What you do in SkirtBrim::generatePrimarySkirtBrimLines is loop through the specified amount of times to do an object offset and turn that into a path. To get what I want, you would need to do the first offset and a clipped version of the last offset and then fill in that zone. It may be a little bit of a change from what you are doing but it might be doable since you do area fills for many other things.

nathanielstenzel, 2021-09-16:
There are a few other options too like still doing a bunch of offsets but adding in some bound checks and either looping back or adding area fills in then or like generating all of the area and then modifying the paths later.

nathanielstenzel, 2021-09-16:
So you seriously can not just ClipperLib::ctIntersection the brim path after it is done being made to keep it in the build-able area?

Ghostkeeper, 2021-09-23:
> I was told the requested adjustment was not possible and I considered a possible alternative

Ah sorry. It is possible, just not very simple.

> So you seriously can not just ClipperLib::ctIntersection the brim path after it is done being made to keep it in the build-able area?

No. Every brim contour would be clipped to the same region and still be polygonal shapes, so you'd get 12 brim lines overlaid on top of each other at the border, resulting in massive overextrusion and blobs that pull the print of the build plate.
The main point is that we need to interpret these brim lines as polylines rather than polygons. It needs to interrupt the material flow and make a travel move in between. This has two problems:
* It's a bit of a refactor. Within the brim generation it's a minor refactor but we'll also have to deal with how we clip the support structure then when it's no longer a closed shape. And of course the disallowed areas in the front-end would need to change.
* It would lend the print to start curling up at the interruption. We don't know how much of an effect that would be on reliability.

We have an issue in our backlog to make this change, albeit for a different feature request: Allowing brims to be enabled/disabled per object. It would also need to allow for brims to be interrupted. We have several feature requests that depend on this change in fact.

MariMakes, 2024-01-11:
Quick update from our side 👋
We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about clipping the brim to the buildplate. 

This functionality was introduced last March in Cura 5.3. 
If you download the latest version of Cura you can start using it.
[image]

I'll be closing this issue since there is a solution in place. 
Thanks for your report. Together we can make Cura better 😊