Ultimaker/Cura #10167: Automatically drop models to build plate option always on / ignored

sgparry, 2021-07-21:
### Application Version

4.10.0

### Platform

Ubuntu

### Printer

Vertex Delta K8800

### Reproduction steps

Start cura
Preferences -> Configure Cura -> General -> Automatically drop models to the build plate: No
File -> Quit
Create a 10mm cube in FreeCAD.
Reposition the cube at z=10mm.
Create a simple copy of the cube to 'bake in' the transformation.
Export the copy to an STL file.
Start Cura
File -> Open
Choose the cube STL File

### Actual results

The cube appears at z=0 on the print bed. No combination of commands seems to reset it to its original position.

### Expected results

The cube should appear 10mm above the print bed

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
These are the files:
[models.zip](https://github.com/Ultimaker/Cura/files/6859263/models.zip)

fvrmr, 2021-07-22:
Hi @sgparry thank you for your report.
This is expected behaviour, since an stl does not save the coordinates from the cad program. An 3mf file does, so if you want to have the coordinates it is best to export your models to 3mf.

The drop automatically to buildplate is for when you move your object in Cura and not with opening a file. I will close this issue and mark it as won't fix, but let me know if this makes sense to you.

closed 2021-07-22 by fvrmr

sgparry, 2021-07-22:
The overall STL model has no origin  coordinates, but the individual vertices do; Cura insists on adjusting these downwards to the base plate regardless of any settings. I appreciate that for most cases this is very sensible, but for certain use cases, e.g. some kinds of multi-part printing - this is a major pain.

fieldOfView, 2021-07-22:
> some kinds of multi-part printing - this is a major pain.

Select the models, and "Merge" them in Cura. They will be repositioned relative to their original origin.