Ultimaker/Cura #10168: line of fillament

jonilma, 2021-07-22:
### Application Version

4.10

### Platform

windows 10

### Printer

FLSUN Q5

### Reproduction steps

Slicing cutters

### Actual results

When printer is hot and starts to print, its start printing a line in the hot bed then it starts printing. This is happening since last version update cause I have print things save with other version and it does not print the line. 
[image]
[image]

### Expected results

No filament lines

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no
[image]

Asterchades, 2021-07-22:
Sounds like a purge line was added to the printer profile in 4.10. There's definitely some customised start code there, some of which draws an arc - and a lot of which is poorly notated. You should be able to manually alter the start GCode in the Printer Configuration to remove or comment out the line starting with G3 and stop it from happening.

Ultimaker doesn't maintain 3rd party printer profiles. They rely on community submissions, and evidently someone in the community decided to add this. The profile author is listed as Daniel Kreuzhofer.

fvrmr, 2021-07-22:
Hi @jonilma your images didn't upload correctly. But it looks like this is indeed a purge line like @Asterchades mentioned which seems to be in your printer profile. 
And this is something we don't maintain, so I will close this issue. Hope you understand. 

But the work around @Asterchades suggested should work for you.

closed 2021-07-22 by fvrmr

jonilma, 2021-07-23:
Hello:

Just open my settings and I don't see any G3 code, please see picture. What can I do?[image]
[image]

[image]
[image]

Asterchades, 2021-07-23:
It's down further in the Start G-code section. Cura doesn't visibly add a scroll bar to indicate that there's more text there (though it will crop the end of longer lines if they're present where such a bar should be), but there should be a fair few more lines under the G92 which you can access by selecting anything in there and using the cursor keys, or just hovering over it and using the scroll wheel. The line that actually draws the arc (G3) should be three lines further down, following an M106 and a G1.

jonilma, 2021-07-25:
@Asterchades thank you very much! I just did it I hope this works, I deleted the line completely. Im new to this so thanks again for your time.