Ultimaker/Cura #10170: Temp Drops after hours of printing

rlcrom, 2021-07-22:
### Application Version

4.8.0

### Platform

windows 10

### Printer

Tenlog Hands 2

### Reproduction steps

I'm at wits end with a problem. I have 5 tenlog hands 2 machines. I'm printing some models that ranges from 3 inches high to 5 or 6 inches tall. The shorter ones print perfect, however, once I get up to, say, 3 inches the prints are failing. I noticed the hot end temp is running at 175 instead of the 205 that it is set for. Same with another machine runs fine to a point then fails and I seen the temp of the hot end trying to print at 160 and in cura I have this to run at 205. Connections to the heads are fine. I'm using Cura 4.8.0. Is there a setting that is causing this? Since it is happening on all my printers? I have had failed prints for a week now and don't know where to go.

### Actual results

Failed print that leaves a pile of spaghettis  or a print that stops adhearing

### Expected results

Perfect print

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[New Compressed (zipped) Folder.zip](https://github.com/Ultimaker/Cura/files/6861396/New.Compressed.zipped.Folder.zip)

fvrmr, 2021-07-22:
Hi @rlcrom thank you for your report.
I noticed in your gcode that around layer 837 the nozzle temperature goes to 150:

```
;LAYER:837
;TYPE:CUSTOM TEMP
M104 S150
```
I also noticed that there stands type:custom temp that indicates that you are probably using a post processing script. But I don't what post processing script because in Cura we don't have a custom temperature script. Could you check if you can disable this script and solves your problem?

rlcrom, 2021-07-22:
Bingo! that was it. Thanks for your time. I was lost on the issue

closed 2021-07-26 by nallath