Ultimaker/Cura #10171: need a solution

sannsint99, 2021-07-22:
### Is your feature request related to a problem?

[image]

### Describe the solution you'd like

hello. i want to know how to remove the outer walls connecting the zig zag lines. i only want zig zags without the walls. help me with this

### Describe alternatives you've considered

-

### Affected users and/or printers

-

Asterchades, 2021-07-23:
There's a setting called "Connect Support ZigZags" which could be causing this. Another option is a "Support Wall Line Count" greater than zero. You may need to adjust setting visibility to be able to see one or both of these.

Also worth noting is that you have a post processing script enabled with one step. Make sure it's appropriate for your current project - I've seen and experienced more than one job doing odd things due to an old script running.

Ghostkeeper, 2021-07-29:
Yeah, this looks like Support Wall Line Count being set to 1, where the user wanted it at 0.

closed 2021-07-29 by Ghostkeeper