Ultimaker/Cura #10172: [BigTreeTech - Firmware/Screens] Flow is not displayed corectly

omexlu, 2021-07-23:
### Application Version

4.9.10

### Platform

Ubuntu 20.04

### Printer

Artillery Genius

### Reproduction steps

1. Sliced with Cura.
2. Start Print.

### Actual results

The flow set in Cura is not shown on the display, e.g. if I set flow to 89%, it still shows 100% on the display.

### Expected results

As with IdeaMaker and other slicers, the BigTreeTech Firmware (Screens)  should then show the set flow. In this case it is 89%, but if it is not it will show 100% throughout. I assume that something is wrong in the gcode for the Artillery Genius in combination with the BigTreeTech firmware / screens . IdeaMaker does this correctly.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing to upload here.

Asterchades, 2021-07-23:
Cura adjusts the E values it writes in the G-Code based on the Flow settings you give it, not the machine flow setting. As such the machine is correctly showing 100% flow because it is using 100% of the value that it is passed.

IdeaMaker instead issues M221 commands. You could (correctly) make an argument that this is a more accurate way to go about it, but unless you're using extremely low flow percentages (sub-10%) you'd be mostly splitting hairs.

nallath, 2021-07-23:
We have to do it like this, because Cura can control the flow per feature (walls, infill, etc). So if we would use m221, the displayed flow might change up and down if you use that.

Instead we simply argue that the entire g-code is changed and that the flow on screen is an extra modifier that can be set by the printer (eg; The printer starts at 100% of whatever the g-code asks).

closed 2021-07-23 by nallath