Ultimaker/Cura #10176: Multi-Extrusion - Mixer extruder, Tool changes enforcing a M104 T command when not necessary.(loss of Shared Heater option) Custom CFFFP

snoopuk, 2021-07-23:
### Application Version

4.10

### Platform

w10

### Printer

Custom FDM printer - Geeeetech A20T

### Reproduction steps

Using Mixer extruder (M163) configure mix ratios and create the printer. 
Set to use third extruder. 
T commands are placed in Gcode, on 4.7 shared heater suppressed this, option no longer seems to exist. 

### Actual results

T2 commands are placed in Gcode and cause issues with OCtoprint and printer errors.

### Expected results

T2 etc should not be placed into the Gcode as set by initial M104 commands.
[cura.log](https://github.com/Ultimaker/Cura/files/6868968/cura.log)
[45be6ad0-c133-479c-a252-d0dd6c521546.zip](https://github.com/Ultimaker/Cura/files/6868975/45be6ad0-c133-479c-a252-d0dd6c521546.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[45be6ad0-c133-479c-a252-d0dd6c521546.zip](https://github.com/Ultimaker/Cura/files/6868981/45be6ad0-c133-479c-a252-d0dd6c521546.zip)

GregValiant, 2021-07-27:
Go to the MarketPlace and load the plugin "Printer Settings".  Within Printer Settings are "Extruders Share Heater" and "Extruders Share Nozzle".

Ghostkeeper, 2021-07-30:
Indeed, this is a print profile issue. The settings can still be enabled as a machine setting (by the printer definition), but you're using the Custom FFF Printer definition which doesn't have that enabled.
The Printer Settings plug-in is a workaround for if the proper printer profile is not available. You can adjust some properties inherent to the printer hardware then, like this one.

There is a Geeetech A20T profile by default in Cura though, which I would expect to work properly. If it doesn't, it'll take a pull request for the Geeetech users to update the default profiles that come with Cura.

This is expected behaviour, based on the settings and profiles you're using.

closed 2021-07-30 by Ghostkeeper

snoopuk, 2021-07-30:
> Indeed, this is a print profile issue. The settings can still be enabled as a machine setting (by the printer definition), but you're using the Custom FFF Printer definition which doesn't have that [...]

The issue is on both original Amit L published profile and custom FFF, the T2 etc commands aren't used with mix extruders in the Geetech A20T Marlin,so even setting it using the M164 you can't actually use it.

the A20T inbuilt profile has ALWAYS set the mix to 
M163 S0 P0.33
M163 S1 P0.33
M163 S2 P0.33

so it always mixes independent of alloocated extruder at a 1/3 - 1/3 - 1/3 mix of filaments. Not ideal.

Ghostkeeper, 2021-08-03:
> The issue is on both original Amit L published profile and custom FFF, the T2 etc commands aren't used with mix extruders in the Geetech A20T Marlin,so even setting it using the M164 you can't actually use it.

Then perhaps it needs an adjustment to the built-in A20T profiles?

Like I said, if it doesn't work out of the box, it takes a pull request for the Geeetech users to update the default profiles that come with Cura. We can't make those changes ourselves since we can't test them with the printer; we'd need to buy every printer in the market to do that.