Ultimaker/Cura #10178: Cura 4.9 and 4.10 fail to complete a slice using custom profile settings

MarkALynch, 2021-07-23:
### Application Version

4.10.0

### Platform

Windows 10 Enterprise

### Printer

Ender 5 Plus

### Reproduction steps

I added parts to the bed, clicked Slice, it gets about 30% and pauses, after 5 or so minutes nothing, but I can click cancel and it stops.

I selected the generic "Draft Quality - 0.32mm" profile and it slices perfectly.
[cura.log](https://github.com/Ultimaker/Cura/files/6870962/cura.log)
[Good .6mm Standard Profile ACCURATE.zip](https://github.com/Ultimaker/Cura/files/6870969/Good.6mm.Standard.Profile.ACCURATE.zip)
I have compressed and included the Zip file of the profile I tired to use and where it failed, tried other profiles that I've created and they fail also.

I have included the STL zip of the items I was going to print. I had them all on the print bed for the 5 Plus.
[Ulvheim+Noble+House.zip](https://github.com/Ultimaker/Cura/files/6870974/Ulvheim%2BNoble%2BHouse.zip)

Mark Lynch

### Actual results

Starts to slice, gets about 30% then the progress bar hangs, then it shows 0%, and it never completes, does allow me to cancel.

I tried this is Prusa and it works fine. I tried this in 4.9 and it did the same hang/fail.

### Expected results

Should complete slicing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I've attached logs and files.
[image]
[image]
[Ulvheim+Noble+House.zip](https://github.com/Ultimaker/Cura/files/6871000/Ulvheim%2BNoble%2BHouse.zip)

fvrmr, 2021-07-26:
Hi @MarkALynch can you post a project file (file - save project) instead of a stl. It is easier to reproduce your issue with a project file.

MarkALynch, 2021-07-26:
Interesting, when I loaded the full stl, the one I created, with all 5 parts on it (they are individual stl) it sliced today. When I clear the build and load the 5 individual STL and try different profiles it gets to about 50% and then the progress clears and shows blank (0%). I have uploaded the project file for the one that doesn't slice.
[Code2_Ulvheim_Noble_House_1A (2).zip](https://github.com/Ultimaker/Cura/files/6877562/Code2_Ulvheim_Noble_House_1A.2.zip)

Ghostkeeper, 2021-07-30:
I'm getting this assertion failure when slicing that project file:

```
2021-07-30 13:23:45,943 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] CuraEngine: /home/trin/Gedeeld/Projects/CuraEngine/src/utils/polygon.cpp:759: static bool cura::ConstPolygonRef::smooth_corner_complex(cura::Point, cura::ListPolyIt&, cura::ListPolyIt&, int64_t): Assertion `vSize(new_p2) < 400000' failed.
```

That indicates that indeed something went wrong internally in CuraEngine.

MarkALynch, 2021-07-30:
For the longest time I had Cura 4.6, 4.7, 4.8, 4.9 and 4.10 installed side by side on my computer. I started seeing things hang/fail to finish slicing in 4.9. I could often go back to 4.7 or 4.8 and slice the same stl with the same profile and it would complete. After installing 4.10 I removed 4.6-4.8 and only have 4.9 and 4.10 installed. I verified that it fails to slice in both 9 and 10. I have seen this happen in a few stls that slice fine in Prusa.

fvrmr, 2021-07-30:
I can reproduce your issue now. Will bring this up for discussion with the team. Keep you posted.

MarkALynch, 2021-07-30:
You guys are amazing - I love Cura.