Ultimaker/Cura #10179: Remove print settings profile override

krikru, 2021-07-23:
### Is your feature request related to a problem?

I want a specific the print setting to be governed by the material print settings, but the print settings profile I'm using has an override for that setting. I want to remove that override, but the only way to do that seems to follow [this procedure](https://community.ultimaker.com/topic/33250-how-to-remove-a-print-profile-setting-override/?do=findComment&comment=288589), which requires temporarily closing Cura and editing the print setting profile file in Notepad.

### Describe the solution you'd like

There should be some way to remove a print setting override directly in Cura.

### Describe alternatives you've considered

No alternatives have been considered.

### Affected users and/or printers

Problem is not specific to any user or printer.

fvrmr, 2021-07-26:
Hi @krikru thank you for your feature request. I will bring it up for discussion with the team. 
Keep you posted!

fvrmr, 2021-07-27:
Hi @krikru it seems that we already got a ticket for this on our backlog and there is a duplicate issue on Github see: https://github.com/Ultimaker/Cura/issues/1820
So I will close this issue as a duplicate.

closed 2021-07-27 by fvrmr