Ultimaker/Cura #10180: Crashes

DocDelete, 2021-07-24:
### Application Version

4.10

### Platform

Windows 10 Build Version 10.0.19042 Build 19042

### Printer

Creality Ender 3 V2

### Reproduction steps

Slice a large object 3" moon.  Go tp preview.  Application crashes.

### Actual results

App Crashes and has to be restarted

### Expected results

Show preview

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6872570/cura.log)
[https://www.dropbox.com/s/ok3khbm63fjbs9j/CE3_Moon.zip?dl=0](url)

fvrmr, 2021-07-26:
Hi @DocDelete does this happen for every model or just this one? Could you try to run Cura in compatibility mode? Go to preferences - configure Cura and enable force layer compatibility mode.

DocDelete, 2021-07-26:
Hi Ultimaker/Cura,
  I am rather new to this.  I have seen it crash before but did not
associate it with one process until now.  I do have 32Gb of RAM in this
PC.  I will try compatibility mode and see what happens.
*Ken Libutti, K4KCL <https://www.qrz.com/db/K4KCL>*

closed 2021-08-10 by no-response[bot]