Ultimaker/Cura #10181: Poor seam alignment

Wasser97, 2021-07-24:
### Application Version

Arachne

### Platform

Win 10, gtx 1550ti

### Printer

Custom, flying bear ghost 5

### Reproduction steps

Upgrade to version 4.10, then upgrade to arachne. The problem appeared already on version 4.10. Completely deleting profiles, cleaning the registry and reinstalling did not help
[image]
[proba100.zip](https://github.com/Ultimaker/Cura/files/6872662/proba100.zip)

### Actual results

The seam is lost in the slicer and when printing

### Expected results

Align the seam in a line

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The files are attached above

smartavionics, 2021-07-24:
For a Cura variant that will align that seam as you desire, see https://github.com/smartavionics/Cura/releases. To get that benefit, you need to set the z-seam alignment to user specified and the seam corner preference to none. Hope this helps...

[image]

fvrmr, 2021-07-30:
Hi @Wasser97 We have a ticket on our backlog related to the seam in Arachne. 
Devs see CURA-8053

For more info about how Cura is place the z seam can you read here: https://github.com/Ultimaker/Cura/issues/8010#issuecomment-654193325

closed 2024-12-09 by github-actions[bot]