Ultimaker/Cura #10183: Cura manages optical output and switches it to bluetooth mode

Lorthiz, 2021-07-25:
### Application Version

4.10

### Platform

Windows 10

### Printer

irrelevant

### Reproduction steps

1) Connetcted optical cable to soundbar with optical/bluetooth modes
2) Start Cura

### Actual results

Cura somehow switches soundbar mode to bluetooth for some unknown reason and by that turns off all PC sounds.

### Expected results

Cura does not change optical output on it's own

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

After switching it back manually using remote it still switches to bluetooth. Problem is gone the moment I turn off Cura and switch it back manualy.

Cura is printing/slicing software why does it mess with optical output?

fieldOfView, 2021-07-25:
If you don't use USB Printing from Cura, you can turn off the USB Printing plugin via the "Installed" tab of the "Marketplace". This will likely stop Cura from accessing hardware it has no business with.

Lorthiz, 2021-07-25:
@fieldOfView Just checked it and you were right it's fixed now. Thank you!
Since this is a plugin problem I am closing this issue.

fieldOfView, 2021-07-25:
This will likely be closed as a duplicate of #6836, which was marked as a duplicate of #5207

fvrmr, 2021-07-26:
I will close this indeed as a duplicate of #5207

closed 2021-07-26 by fvrmr