Ultimaker/Cura #10184: Z coordinate value always reset to zero after saving the project to a .3mf file.

twinclouds, 2021-07-25:
### Application Version

4.6.1, 4.9.1, 4.10.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Change the z-coordinate to a non zero (nagative) value.
save the project to a .3mf file.
Close Cura and restart.
Check the new z-coordinate value.

### Actual results

The z-coordinate value always becomes to 0.

### Expected results

The z-coordinate value should be the same before saving.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

That's it.
[image]

fvrmr, 2021-07-26:
Hi @twinclouds thank you for your report. I can't reproduce your issue, tried it with grouped models and support blockers. Does it happen with every model? Could you share your project file?

twinclouds, 2021-07-26:
Hi, Fanny:I checked a couple of models.  For 3DBench, the z-coordinate change saved fine.  However, the problem happens to most of my models and every version of Cura since 4.6.1. I am not sure it has anything to do with grouped or not.  I have attached link to a 3MF.zip file containing the original bull-s.3MF file and the CE3PRO_Bull-s.3MF file saved after shifting z-axis by -0.5mm.  The results are the same, i.e., the shift of -0.5mm got lost.  Could you take a look?Link: 3mf.zip

| 
| 
|  | 
3mf.zip

 |

 |

 |

Thank you very much!Fuyun

fvrmr, 2021-07-27:
Hi @twinclouds I can't see your attachment since you replied via email. Could you post it as a zip here on Github?

twinclouds, 2021-07-27:
Actually, I attached a link and you can see from there.  Here is the link again: (Link: 3mf.zip).  I will try to upload on github but I haven't do that before.Fuyun

twinclouds, 2021-07-27:
Here is the same link.  Hopefully you have no problem to open it.
[https://drive.google.com/file/d/1B75A1Jh3T-1_nlTyRBlLWkvUx0txy3pH/view?usp=sharing](url)
Thank you!

twinclouds, 2021-07-27:
Here are two files.  The file with "shift" is the one saved after shifting z by 0.5 mm.  The shift was not saved.
[Bull-s.zip](https://github.com/Ultimaker/Cura/files/6887691/Bull-s.zip)
[Bull-s-shift.zip](https://github.com/Ultimaker/Cura/files/6887698/Bull-s-shift.zip)

twinclouds, 2021-07-28:
Hi,Hope you have received my files this time.  Thank you for your help!Fuyun

GregValiant, 2021-07-28:
Here is how bull-s-shift.3mf it came in for me (Z = -0.5).   Bull-s came in at Z = 0.
[image]

twinclouds, 2021-07-28:
Hi, Thank you for getting back to me.  This is what I got from Cura v4.10.0 and W10.  Which version of Cura and platform (OS) are you using?  

Could you tell me you think what happened?Thanks.Fuyun

twinclouds, 2021-07-29:
Is there anything I did wrong?Fuyun

GregValiant, 2021-07-30:
Win 10 and Cura 4.10 and an Ender 3 Pro.
There have been some complaints lately about things moving in the Z when a 3mf file is opened.  I think that so far they have been tough to duplicate.  It's like trying to chase down an intermittent automobile electrical problem.  It drives you crazy so you take it in to the repair shop and it refuses to act up for the mechanic.

twinclouds, 2021-07-30:
No.  It's not only for Cura 4.10.  I have tried 4.6, 4.8 they all have the same problem.  In addition, it has nothing to do with what 3D printer that I am using.  Could you please tell me which version on which platform the saving works?  I will be happy to try it if convenient.  Actually, the x and y coordinates that I saved are both zero but what you had shown are not.  So even your version did something incorrect.  I don't this is an unrepeatable case as you discussed.  Hopefully you guys can figure out a way to determine what this problem is.
Fuyun

fvrmr, 2021-07-30:
Hi @twinclouds do you write it down manually? Because it could be a duplicate of https://github.com/Ultimaker/Cura/issues/10100
For me if I use a `,` it doesn't save the value.
Could that be that be the case for you?

twinclouds, 2021-07-30:
Hi, Fanne:  Not sure what do you mean by "write it down manually".  In any case, I was taking about the coordinate changes.  Not about scaling changes.  Fuyun

twinclouds, 2021-07-31:
I tried to use "," instead of ".".  It didn't take "-0,5" as a valid input for "-0.5", so no shift was performed in the first place.

fvrmr, 2021-08-03:
>  Not sure what do you mean by "write it down manually". 

Typing the coordinates in the input box in the move tool, instead of dragging it with the arrow. 

Did you see that your model was moved before you saved your project?
And does it happen with other models.

twinclouds, 2021-08-03:
Hi, Fenne:Yes, the coordinates are typed in, not by drawing, and I checked them every time before I save the project.It happen to most of my models.  I did find that the 3Dbench saved correctly.Do you find my model of bull saved correctly in your setting?  If you do please let me know the conditions that I can try to repeat.Thank you!Fuyun

Ghostkeeper, 2021-08-05:
In the bull-s-shift file, it stored it as moved 0.5mm down. That part is working fine.

The issue at hand here is the "Automatically drop models to the build plate" preference. If enabled, Twinclouds' symptoms reproduce. If not, the model loads in 0.5mm lowered into the build plate.

I'll confirm that this is a bug. Although you could interpret it both ways... It's aligning the bottom of the model to the build plate properly here, but that's clearly not what the user wanted. If the file would have models floating up in the air, it's fair to say that most likely the user wants to drop them down. But if it is lowered underneath the build plate, the user most likely intended it to be lowered in order to cut off the bottom side.

As a workaround, temporarily disable the "Automatically drop models to the build plate" preference when loading in this model.

twinclouds, 2021-08-07:
Thanks for finding out the root cause of the problem.  I disabled "Automatically drop models to the build plate" and it works fine.  I am satisfied with this solution, even though I hope that a permanent fix as you suggested can be implemented in the future revisions.Fuyun

fvrmr, 2021-08-12:
I have created a ticket on our backlog for this bug. 
Devs see CURA-8479

closed 2024-12-09 by github-actions[bot]