Ultimaker/Cura #10185: Material settings build plate temperature does not update print settings

whyme12, 2021-07-25:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Modded Anycubic i3 Mega

### Reproduction steps

Change material from PLA to ABS and compare the buld plate temperature with the temperature in the material profile.

### Actual results

As seen in the attached picture, the buildplate temperature was not changed when selecting a new material.

I have checked the gcode, it is not a visual bug, the parameter is not updating.

### Expected results

All settings of the the material profile should changed when selecting a new material.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-07-26:
Hi @whyme12 I can't reproduce your issue.
[image]

Could you share your definition file of your printer? It is possible that it is defined there. You can find your definition here: `C:\Program Files\Ultimaker Cura 4.10.0\resources\definitions`

whyme12, 2021-07-26:
Found the problem, bed temperature was stored inside the quality settings ("...AppData\Roaming\cura\4.10\quality_changes").

But not sure when and why I should have changed it.

closed 2021-07-26 by whyme12