Ultimaker/Cura #10191: Feature Request: radial infill, for printing circular objects

gitcnd, 2021-07-26:
### Is your feature request related to a problem?

There is no existing infil pattern that is sensible for circular objects (e.g. a hollow pole)
[image]

### Describe the solution you'd like

For items like these, the correct infill pattern needs to be lines that radiate out from the center - like an asterix *

Ideally, allowing these radiating-out lines to be optionally rotated slightly per layer, and also optionally allowing two sets of these (rotating in opposite directions) would be perfect - this would give great inner-to-outer wall connectivity with some added 3D support.

### Describe alternatives you've considered

I tried all the existing patterns - they all waste heaps of filament drawing parallel to existing walls, instead of connecting them together. (see screenshot)

### Affected users and/or printers

Everyone making hollow circular objects will benefit.

### Additional information & file uploads

As far as I know - no other sliders do this either - so this would be a great feature.

gitcnd, 2021-07-26:
* maybe "snowflake" is a better analogy than asterix - you get the idea though - the inner and outer walls need to be connected with 360-degree symmetry, without wasting time adding pointless extra infill lawyers to existing walls.

gitcnd, 2021-07-26:
[image]

fvrmr, 2021-07-27:
Hi @gitcnd thank you for your feature request.
It seems to be related to this issue: https://github.com/Ultimaker/Cura/issues/8811 but there was not a decision made about this. So I will discuss this with the team.
Keep you posted!

rburema, 2021-07-28:
Per policy, discussion will continue in #8811 in order to not have two separate open issues on the exact same topic. As such, this will be closed, as all duplicates are.

As the issue is more complicated that many people seem to realise, we'll have a more technical meeting either today or next week on whether to continue to investigate this, and if so, how. The conclusion will be posted at that time to the other thread.

closed 2021-07-28 by rburema