Ultimaker/Cura #10197: Cura close itself after loading

acavallo-thux, 2021-07-27:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Can't choose Printer

### Reproduction steps

1.Start Cura
2. After loading , the program close itself

### Actual results

Cura close the program

### Expected results

Start Cura
[cura.log](https://github.com/Ultimaker/Cura/files/6884367/cura.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

2 monitor 
Nvidia 1070ti Video Card
Samsung G7 Monitor 240Hz

Tried all the solutions:
deleted cura folder in appdata\roaming
compatibility mode
disconnected monitor 1
disconnected monitor 2
uninstalled and reinstalled

fvrmr, 2021-07-27:
Hi @acavallo-thux thank you for your bug report. 
I will bring this up for discussion with the team. Keep you posted!

Ghostkeeper, 2021-07-28:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/7744 . There are a couple of workarounds in that thread you can try.

This issue, by its nature, is difficult to solve. The graphics card is telling Cura that it supports OpenGL 4.1, but then doesn't support the shader language GLSL 4.1 which is an integral part of OpenGL 4.1. It's like telling a contractor you can build a house, but then not telling him how you don't have any idea how to build a wall until after the contract is signed.

closed 2021-07-28 by Ghostkeeper