Ultimaker/Cura #10199: When I use 4.10, I can't get the fill shown in Figure 2,

coolgua888, 2021-07-27:
### Application Version

4.10.0

### Platform

win10

### Printer

tronxy xy-2 pro

### Reproduction steps

Model wall thickness is 1mm
When I use 4.8, I get the fill shown in Figure 1,
When I use 4.10, I can't get the fill shown in Figure 2,
4.8 print thin wall filling mode is more stable, 4.10 will be divided into two layers.
When using padding, is it possible to change 4.10 to the same as 4.8?
4.10 also has a bug, sometimes I click on some menus, still stuck, the computer will become very slow.
My computer has a 16G flash memory i5CPU

### Actual results

In 4.10, the walls of the model will be divided into two vertical layers.
4.10 will get stuck

### Expected results

fine
[image]
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fvrmr, 2021-07-29:
Hi @coolgua888 not all your images are uploaded. Can you edit your post and post them again. Could you also share a project file where your issue is happening. In that way it is easier for me to reproduce this as well.

Ghostkeeper, 2021-08-03:
The pattern shown in the picture under "expected behaviour" is not expected behaviour. Those cross-sectional infill lines are overextruding. They should've been removed by the gap filling routine. So I think this is not a bug.

closed 2021-08-12 by no-response[bot]