Ultimaker/Cura #10200: programm can't slice gcode from stl

Svarozhez, 2021-07-27:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creality CR-10S

### Reproduction steps

1 download the cura v 4.10
2 install cura v 4.10 without removing old working version(4.6)
3 set needed for slicing settings
4 insert file in cura v 4.10
5 try to slice gcode 

### Actual results

programm write that it is slicing, but it goes near 30 minutes, untill this time i open this file in cura v 4.6 and it generate gcode in 2 or 3 minutes (gcode on 446 Kb)

### Expected results

i expect that cura v 4.10 slice gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

screenshot of problem(selected in right down corner)
 [image]

[cura.log](https://github.com/Ultimaker/Cura/files/6887526/cura.log)

in archive loaded original stl file, generated by cura v 4.6 gcode and project file of cura v4.10
[3mf stl gcode.zip](https://github.com/Ultimaker/Cura/files/6887536/3mf.stl.gcode.zip)

GregValiant, 2021-07-27:
Show all settings and then look in Support.  You see three settings whose values are in orange boxes.  Orange means a warning. 
 In your case set "Support Density" to anything greater than 0.
This might be a bug as it appears the box should be red which would indicate that Cura can't slice.  It might be because you are asking for support and then setting it to 0% density which would mean no support.

Ghostkeeper, 2021-08-03:
The problem is with the Cross pattern used for Support here.

It ends up in an infinite loop here:

https://github.com/Ultimaker/CuraEngine/blob/2bd988bd79446b9d41b6a6c33626f71f7d92d77b/src/infill/SierpinskiFillProvider.cpp#L73-L77

That happens because `aabb_size` starts off 0, and no matter how many times you double 0 it will never get above `max_side_length`. Which is because `min_line_distance` is given as 0. Which is because the Support Line Distance setting is 0. So the conclusion is that the Cross support pattern doesn't deal well with infill density 0.

The pattern itself is not generated if the infill density is 0. But there is a pre-compute stage that is being called regardless. Preventing that from happening with infill density 0 makes sense.

This issue is fixed here: https://github.com/Ultimaker/CuraEngine/commit/a0b20d8fbe29913b93b4410212471b2847af8ea3

closed 2021-08-03 by Ghostkeeper