Ultimaker/Cura #10201: Bug Multiply Objects

andmuniz, 2021-07-27:
### Application Version

Cura Arachne_engine_beta

### Platform

Windows 10

### Printer

Anycubic mega S

### Reproduction steps

When slicing process **many objects** the process stop. 

### Actual results

Slicing process stop. 

### Expected results

Slicing many objects.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6889227/cura.log)
[image]

fvrmr, 2021-07-29:
Hi @andmuniz can you also share your project file? I can't reproduce this issue myself.

andmuniz, 2021-08-02:
> 
> 
> Hi @andmuniz can you also share your project file? I can't reproduce this issue myself.

[AI3M_teste.zip](https://github.com/Ultimaker/Cura/files/6913705/AI3M_teste.zip)

fvrmr, 2021-08-03:
Hi @andmuniz thank you for your project file. I can reproduce your issue and it seems like this is caused by some changed settings because if I change to the default profile, I can slice your project. 
[image]

I will bring this up with the team. Keep you posted.

andmuniz, 2021-08-03:
> I will bring this up with the team. Keep you posted.

Okay, thank you... and i appreciate that I can contribute in any way.

Ghostkeeper, 2021-08-09:
Specifically, it seems that the infill density of 15% instead of 25% seems to be causing this error. Looks like an edge case.

konskarm, 2021-08-16:
Hi @andmuniz. My apologies for the late response. This issue seems to be happening somewhere in the InsetOrderOptimizer. We already have an internal ticket for that. Devs, see CURA-8106.

Piezoid, 2021-08-24:
Hopefully this bug should be solved with [PR #1487 (CuraEngine)](https://github.com/Ultimaker/CuraEngine/pull/1487)

Ghostkeeper, 2021-09-03:
Can confirm! This bug is indeed fixed now. Thanks for your help, Piezoid!

closed 2021-09-03 by Ghostkeeper

andmuniz, 2021-09-07:
> 
> 
> Hopefully this bug should be solved with [PR #1487 (CuraEngine)](https://github.com/Ultimaker/CuraEngine/pull/1487)

Hello, But I didn't undesrstand how apply this solution in my software.

fvrmr, 2021-09-09:
You can run Arachne from source, read here how: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source
Other than that you need to wait till we release a other beta of Arachne.