Ultimaker/Cura #10203: Can't resize "Save to Disk" dialogue

krikru, 2021-07-29:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creality Ender-3

### Reproduction steps

1. Start Cura
2. Import a model by clicking-and-dragging an stl file into Cura
3. Slice the model
4. Click "Save to Disk"

or

1. Start Cura
2. Import a model by clicking-and-dragging an stl file into Cura
3. Click "File"->"Save Project..."
4. Click "Save"

Then try to resize the dialogue window that is opened.

### Actual results

The dialogue window won't be resized.

### Expected results

The dialogue window should be resized.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

fvrmr, 2021-07-29:
Hi @krikru I can't reproduce your issue. Could you make a screencast of what is happening.
Here you can see it is working for me: 
[image]

Ghostkeeper, 2021-08-05:
Perhaps this is something specific to your operating system. Cura doesn't create that dialogue. It calls upon Qt which calls upon your operating system to create that dialogue.

closed 2021-08-12 by no-response[bot]