Ultimaker/Cura #10205: Differing layer heights for infill and walls

Juusoz, 2021-07-29:
### Is your feature request related to a problem?

There is a method to improve print time without losing quality.

### Describe the solution you'd like

I have a suggestion that could effectively cut the print time in half for some high-detail prints.

**Just print the outer walls with a different layer height from the rest.** It would allow the print quality to look like it was printed at 0.1mm layer height, but with the print time being close to a 0.2mm layer height.

The supports can also be printed at a multiple of the layer height (which it already is, _Support Infill Layer Thickness_), but the support roof should be printed at the same layer height as the outer walls.

[image]
The inner walls are a bit tricky. On one hand, you could totally print it at a multiple of the outer wall height, but that could result in imperfections on the outer wall. And on the other hand, if you print it at the same layer height as the walls, it adds to the print time.
My suggestion would be to print the first inner wall at the same layer height as the outer wall with all subsequent walls with a multiple of the outer wall height. You could, of course, make this user controlled.

### Describe alternatives you've considered

_Adaptive layers_ are fun and all, but it leaves a slight texture change in the affected areas.

### Affected users and/or printers

It should benefit everyone.

Asterchades, 2021-07-30:
Infill can already be printed at a multiple of the global layer height. There's an option called "Infill Layer Thickness" which corresponds to the support setting you're already familiar with.

Juusoz, 2021-07-30:
> Infill can already be printed at a multiple of the global layer height. There's an option called "Infill Layer Thickness" which corresponds to the support setting you're already familiar with.

Oh, so I'm just suggesting a feature that already exists? My mistake.

Kind of wish they would move all the layer height settings to the same place instead of being seperated like that.

closed 2021-07-30 by Juusoz

Ghostkeeper, 2021-08-05:
We could do that, but then there'll be some other guy that wishes all the infill settings were moved to the same place instead of being separated like that ;)