Ultimaker/Cura #10208: Application Freezes Computer

jjjohnsen, 2021-07-29:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Wanhao 6

### Reproduction steps

Brought in STL file and set custom print settings.  Went back to default print settings and selected 'Normal-1.5mm' and hit Slice.  Every program open on my computer stopped working and froze, then computer froze.  Had to do a hard shutdown and restart.  Repeated the same thing with the same file and everything froze again.  Cura file attached.

[MT-R24-Segment.zip](https://github.com/Ultimaker/Cura/files/6903675/MT-R24-Segment.zip)

### Actual results

Computer freezes.

### Expected results

Cura should slice the model.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6903719/cura.log)
[MT-R24-Segment.zip](https://github.com/Ultimaker/Cura/files/6903721/MT-R24-Segment.zip)

fvrmr, 2021-07-30:
Hi @jjjohnsen I can't reproduce your issue, does it happen with other models as well? And does it happen when there are no other programs open?

jjjohnsen, 2021-07-30:
The second time it happened there were no other programs open.  Just tried with another model and yes, it froze the computer again.  Seems like it spikes the CPU to 100% and can't compute the slice.  Not sure I want to keep trying to repeat it again and freeze everything.  I'll just stick with the custom settings.

fvrmr, 2021-08-03:
So your computer doesn't freeze when you use the custom settings?

Ghostkeeper, 2021-08-05:
Do you happen to have an Asus motherboard and a Ryzen CPU? You could be hitting this issue: https://github.com/Ultimaker/Cura/issues/5265

jjjohnsen, 2021-08-06:
> Do you happen to have an Asus motherboard and a Ryzen CPU? You could be hitting this issue: #5265
I have an HP Z420.  Not sure the moterboard, but the CPU is Intel Xeon E5-1650 v2

> So your computer doesn't freeze when you use the custom settings?
No.  It seemed to only happen when switching from the custom profile to the default and slice (I was working with and adjusting custom settings and switched to compare).   Something suspicious with the slice.  
Custom profile attached.
[jjj-custom-01.zip](https://github.com/Ultimaker/Cura/files/6942268/jjj-custom-01.zip)

Ghostkeeper, 2021-08-06:
Okay, so not the Asus freezing issue. Lucky then for you, because that's a really difficult one to solve.

Curious. So something in those custom settings "fixed" the issue. The profile contains like 50 setting overrides though, so it's pretty hard to narrow down! It seems like to complete this slice with the "Normal (0.15mm)" profile, Cura needs about 800MB of free RAM in peak. Other than that it's hard to tell what's going wrong for you that could be causing a complete computer freeze.

We haven't been able to reproduce the problem yet. I hope that other people that run into this issue will report their findings here too so that we can narrow down under what conditions this problem occurs.