Ultimaker/Cura #10209: Not possible to slice objects whose bottom has been lowered to z<0

sixtyfive, 2021-07-30:
### Application Version

4.10.0

### Platform

Linux

### Printer

Custom (highly modified Ender2)

### Reproduction steps

- I opened an .stl file
- I moved it downward so that the object's bottom's location on the z axis became negative
- then I tried to hit the "slice" button

### Actual results

Cura says "Unable to slice".

### Expected results

My expectation was that Cura would ignore everything in and underneath of the build plate and slice everything on and above the build plate.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

That's it, see above. Googling around, I found in #2147, that what I (intuitively) tried to do, used to work.

fvrmr, 2021-08-02:
Hi @sixtyfive I can't reproduce your issue. If I move a model below the buildplate, it does slice. 
[image]

A project file (file - save project) would be help to see if I can reproduce your issue.

fieldOfView, 2021-08-02:
I think that OP uses a model that does not fit the build volume horizontally, and expects Cura to be smart enough to be able to slice the "tip of the iceberg" as it were. The check if the model fits the buildvolume horizontally is always done on the entire model, regardless of if the wide part is under the buildplate.

sixtyfive, 2021-08-02:
> I think that OP uses a model that does not fit the build volume horizontally

No, it does fit horizontally. The lower 95% printed fine, but then it turned out that the printer's build volume is not as large vertically as it says on the label. So had to stop the print and was trying to print the remaining 5% on their own. I'll attach a project file.

sixtyfive, 2021-08-02:
Hah, nevermind, found the problem when I just tried again. It turns out @fieldOfView wasn't wrong, but also not entirely on-spot. Originally, I had first lowered the object on the z axis, but then also wanted to place "the tip of the iceberg" at x/y center. That of course caused "the iceberg" to not be within x/y bed size confines anymore, hence the error message. In that case it's not a bug, of course, but perhaps an opportunity for a bit of a more verbose error message?

Ghostkeeper, 2021-08-05:
Cura can't detect the difference between whether it's going out of the build volume above the build plate or below the build plate, at least not without slicing it first.

We can check the bounding box of subsets of vertices of the model. But leaving out all vertices below the build plate is not a solution. For instance, if you want to print a pyramid with just 5 vertices, partially below the build plate, leaving out everything below the build plate leaves just a single point with no surface at all. Even a truncated pyramid would cause Cura to find too small of a collision area.
Too small of a collision area allows the user to place the model too close to the edge, causing the g-code to exit the build volume. This is dangerous for some printers, which can be damaged by it.

I'll leave this issue open since it is a genuine problem, but solving it is complex enough that I'll mark it as deferred. It's not likely that we'll get around to fixing this soon.

MariMakes, 2023-06-14:
Hi @sixtyfive,

Hooman reporting for duty 👋 
We have so many issues to go through that we have a bot helping us out. It's also there to help us to introduce old tickets to new more efficient procedures. 

Ah the "tip of the iceberg" issue, that's indeed still present and tricky to resolve.
Plug-ins that can cut [models like this one:](https://hci.rwth-aachen.de/m3-autoparter) can make your life easier. I've also had some good experiences using TinkerCad for dividing a model to fit it on the buildplate, as long as the model is not too detailed. 

Would you still like to keep this issue open or do you agree that we can close it? 
It's a tricky problem like @Ghostkeeper suggested, if you would like us to keep it open, what's the best way I could help here?

MariMakes, 2023-06-27:
Thanks for the update @sixtyfive 😊 
In that case I'll close this ticket. 

Happy printing and good luck with the plugin!

closed 2023-06-27 by MariMakes