Ultimaker/Cura #10211: ENDER 6 ADD TO CURA

SSAA308, 2021-07-31:
### Is your feature request related to a problem?

MY ENDER 6  is not in cura ultimaker its very annoying to calibrate to make prints work some times

### Describe the solution you'd like

please add ender 6 to your settings

### Describe alternatives you've considered

none 

### Affected users and/or printers

ender 6 it will work better

GregValiant, 2021-07-31:
Creality (or a user) has to submit a definition file to the Ultimaker Cura team.
I put the attached one together.  

Extract the files to a folder of your choice.
Copy just the files to the corresponding sub-folder in the Cura resources folder.
Since the definition is not installed with Cura you would need to do this again if you update Cura.  I would expect that by the next Cura update Creality will have submitted a definition file for the E6.

[Ender 6 Definition Files.zip](https://github.com/Ultimaker/Cura/files/6911103/Ender.6.Definition.Files.zip)

fvrmr, 2021-08-02:
Hi @SSAA308 we don't maintain these profiles, like @GregValiant mentioned. And we are depend by the community for committing these profiles. 
I will close this issue, hope you understand.

closed 2021-08-02 by fvrmr