Ultimaker/Cura #10213: Gcode

Julwyn, 2021-07-31:
### Application Version

4.10

### Platform

windows 10

### Printer

ender 3 pro

### Reproduction steps

i put my own start G-code

### Actual results

and dont over ride it...

;FLAVOR:Marlin
;TIME:2692
;Filament used: 9.20187m
;Layer height: 0.54
;MINX:72.9
;MINY:67.394
;MINZ:0.4
;MAXX:131.35
;MAXY:152.595
;MAXZ:30.1
;Generated with Cura_SteamEngine 4.10.0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
;My start g-code;;;;;;;;;;;;;;;;;;;;;;;;;;;
M82 ;absolute extrusion mode
;*** Start Dual Nozzle/Bed Preheating ***
M140 S60 ; start preheating the bed
M104 S200 ﻿T0 ; start preheating hotend
M190 S60 ; heat to Cura Bed setting 
M109 S200 ﻿T0 ; heat to Cura Hotend
G28 ; home
;*** End Preheating ***
;-------------------------------------------------
;*** Single Print Start Tone
M300 S1000 P500 ; chirp to indicate starting to print

### Expected results

overide the default start g-code with my own

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

GregValiant, 2021-08-01:
Change your start gcode to this:
;*** Start Dual Nozzle/Bed Preheating ***
M140 S{material_bed_temperature_layer_0} ; start preheating the bed
M104 S{material_print_temperature_layer_0} ﻿T0 ; start preheating hotend
M190 S{material_bed_temperature_layer_0} ; heat to Cura Bed setting
M109 S{material_print_temperature_layer_0} ﻿T0 ; heat to Cura Hotend
G28 ; home
;*** End Preheating ***
When Cura sees the Keywords (in curly brackets) it will replace them with the numbers in your setup and won't put the preliminary temperature lines in.  If you were to hard code temperature lines in the startup gcode (M104 S200) then Cura will put the preliminary stuff in as it can't tell if you are using the Cura settings or just making stuff up.

fvrmr, 2021-08-02:
Hi @Julwyn can you let us know if the solution from @GregValiant works for you.

closed 2021-08-16 by no-response[bot]