Ultimaker/Cura #10214: Losing icons to black squares

Dragonstuff83, 2021-08-01:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

All of the icons are black squares again. I followed the process documented [here](https://github.com/Ultimaker/Cura/issues/1722) where it tells me to go in a delete the QT_PLUGIN_PATH. That fixed it for me, then it randomly happened again, so I deleted the program and reinstalled it. 

Now, today it happened again. Is there a way to make sure this doesn't keep happening?

### Actual results

Black squares

### Expected results

Not black squares.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

fvrmr, 2021-08-02:
Hi @Dragonstuff83 did you delete the `QT_PLUGIN_PATH` also after you did a reinstall?

Ghostkeeper, 2021-08-09:
Cura's installer does not change the `QT_PLUGIN_PATH` variable. Arguably, nobody should ever change it, except a developer working with multiple installations of Qt. But some applications out there on the internet still do, I'm guessing.

The randomly happening again may be because you installed one such application that changed it. This would cause all Qt-based applications to break, including Cura.

closed 2021-08-16 by no-response[bot]