Ultimaker/Cura #10215: Monitor Screen never updates

TRLBY, 2021-08-01:
### Application Version

4.9.0

### Platform

Windows 10P

### Printer

Ultimaker S% Bundle

### Reproduction steps

Run a print... and go to Monitor and watch as it falls asleep!  the only thing that is updated in the Camera. The time and percentage completed is useless.

Especially since it's not even showing the correct part, but the one from the previous print job.

[image]
[image]

### Actual results

The printer says I have almost 2.3 hours left on my print, as does the "Digital Factory" from Ultimaker's website, yet on the Monitor view it says it'll be done a 6:26 today... 

### Expected results

Proper "Monitoring" of the process... if not why not call it "Lucky Guess" or "Come back later when I wake up"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fvrmr, 2021-08-02:
Hi @TRLBY I will bring this up with the team. Keep you posted!

ayuhu, 2021-08-07:
I was just about to open a post as well.
It seems after opening the DF the website refreshes again for about 2 mins. 
After that there are no more website refreshes.

fvrmr, 2021-08-09:
Hi @TRLBY it looks like your connection is not via the cloud. 
See here how to connect to the cloud: https://support.ultimaker.com/hc/en-us/articles/360012019239-How-to-connect-your-printers-to-the-Digital-Factory#h_01F2R86T4K0427GYFDP1Z7T1N6

Ghostkeeper, 2021-08-09:
Indeed, but it's also expected to update via LAN, no?

TRLBY, 2021-08-09:
We had a Power failure at 4:40 PM yesterday and our Power was just restored 30minutes ago, so I DO believe my Printer is on the cloud. But this brings up the another annoying thing about Ultimaker... why am I here complaining about something Ultimaker should be in charge of? If I bring these issues up I'm sent here.... that has got to be the dumbest Business decision I've ever heard of. 

And I don't like having my printer or anything "on the cloud" just to view my printing. 

I know all the Hobbyist love this crap, but I give Ultimaker no hope of capturing serious businesses with this approach. They spend all this time and energy developing their Professional Service to manage multiple machines for $600 a month and leave their CURA software development to a 3rd party. I can't even manage one system let alone many with poor implementations like this.

I don't even know "who's cloud" my Machine is on and what security is in place. We do work for the Government and Fortune 100 companies, and I refuse to put their products up on "the cloud".

Ghostkeeper, 2021-08-09:
Cura is being developed by Ultimaker. Both fvrmr and I are employees of Ultimaker. Instead of complaining here about it, you might as well have asked this to the Ultimaker support channels. They'll be able to help with the most common issues anyway, and will raise the bugs to us if they are bugs (or the cloud team if it's a bug concerning stuff that runs in the cloud). For uncommon or difficult issues they should refer back to us, the Cura team of Ultimaker, as last line of defence. So here you are. I guess they didn't immediately forward it because the Github issue template asks for the necessary information for us to reproduce the issue, such as a log file. A log file would be handy to reproduce this issue, but you didn't provide it yet.

> And I don't like having my printer or anything "on the cloud" just to view my printing.

Indeed, but your printer also supports LAN so that should work just as well. It's not likely to get many new features but it's still maintained. It should update during the print. Apparently for you it doesn't. So something is going wrong there.

> I don't even know "who's cloud" my Machine is on and what security is in place. We do work for the Government and Fortune 100 companies, and I refuse to put their products up on "the cloud".

Ultimaker's cloud services are hosted on Google Cloud. Our software (cloud as well as Cura) is audited by 3rd party security researchers and we have no major security vulnerabilities known at the moment.

TRLBY, 2021-08-09:
Thank you Ghostkeeper for confirming my experience. Not sure why you're defending a practice that you kindly explain as being dysfunctional. 

  "They'll be able to help with the most common issues anyway, and will raise the bugs to us if they are bugs (or the cloud team if it's a bug concerning stuff that runs in the cloud). For uncommon or difficult issues they should refer back to us, the Cura team of Ultimaker, as last line of defence".

Maybe they need a better "defense" instead of da FENCE, which must be a Freudian slip.

Anyway I had hoped Ultimaker was serious when the developed the S5 Bundle, but I guess the "culture" there is still focused on the Hobbyist.  But my guess is with "emploees" like you, I'm not very hopeful you understand what Businesses expect from vendors and suppliers.

Ghostkeeper, 2021-08-09:
> Maybe they need a better "defense" instead of da FENCE, which must be a Freudian slip.

It's just British English, which is what I'm more used to :wink:

I'd love to see your Cura log, still. Since the issue doesn't happen with our own printers, it would help to see what's going wrong for your printer specifically. You can find the log in `%APPDATA%\cura\4.9\cura.log`.

nallath, 2021-08-09:
> But my guess is with "emploees" like you, I'm not very hopeful you understand what Businesses expect from vendors and suppliers.

I understand that you're frustrated that it's not working, but I would kindly ask you to refrain from snarky comments like this. We are trying to help you out here after all :) 

If you feel that we are unable to assist you with your problems, we can get you in contact with people who are trained sales representatives (instead of engineers, which is what all of the people here are).

TRLBY, 2021-08-09:
Nallath,

I would love to talk to "trained sales representatives" but they keep sending me here. So being  "snarky" seems to be my only hope of getting answers.

And I swear if I have to upload my "log" files one more time to get nowhere, I think I scream. Doesn't Ultimaker have a log of all my "uploads"?

And just so you know, your friend Ghostmaker has such a professional sounding name, that I assumed I was dealing with an some kid in their mothers basement and NOT an Ultimaker Engineer.  

But hey I'm being lectured about not acting "professional".

Ghostkeeper, 2021-08-09:
Github being anonymous, we don't have any way to connect your issue with logs you might've given before to Ultimaker's support channels.

TRLBY, 2021-08-09:
Ghostkeeper you keep answering my initial complaint with the same time wasting requests. You don't have my logs, but you say you work for Ultimaker... but then "Github being anonymous, we don't have any way to connect your issue with logs"

So just so we're on the same page:

 Github is NOT part of Ultimaker, and CURA even though owned by Ultimaker, isn't responsible for the problems I'm having with CURA...  and "Github being anonymous"  But please upload your LOG file one more time anyway, to this 3rd party website.

So please, either you or Nallath, tell me why I shouldn't be confused and annoyed at this ridiculous arrangement as a customer.

nallath, 2021-08-09:
We work for Ultimaker. Both Cura and the Ultimaker machines are made by the same company (Ultimaker). There is no third party involvement here what so ever (apart from using products from other people and the internet as infrastructure)

Since we don't know your real name, we don't know if you logged something in the other support systems that we have. Aside from that; most people can't even access those systems, since we have proper separation of concerns and "least rights required" policies in place to ensure the security / privacy of the data that we handle. As an experienced business user that holds security in high regard, I assume that this should not be new to you. 

We need the logs in order to help you. They are not sent to us automatically, so you need to send them to us. If you don't want to do this publicly, you can send them to anyone that has responded here directly.

TRLBY, 2021-08-09:
Again,  Like Ghostkeeper you keep explaining these things to me but only convince me more and more how dysfunctional Ultimaker is. 

So based on your description of how this all works, this is how I understand it:

I as a customer have to use "Github" a 3rd party service in which I have no relationship with for a product made by Ultimaker, but doesn't even know my name, and I'm suppose to think that's a good thing for security reasons, right?

Since I wasn't aware of all these different groups who can't talk or share data with one another due to this extraordinary security, and it's "least rights required" policies, but never informed their "new customers" not to use any identifying names to avoid exposing who they are to these 3rd parties and their platforms. 

CURA which is owned by Ultimaker but uses a different software developer group different then Ultimaker's firmware group, also doesn't know me as a customer again for security reasons.

And when I contact Ultimaker with my issues and problems, they direct me to Github who once again I have NO relationship with, for this fantastic  customer support, then get 2 Engineers who have no idea who I am try to help me.

...Got it! 

So will either of you be able to tell me why CURA and the Ultimaker Digital Factory aren't updating each other? The Folks at Ultimaker who I contacted via email know me, but don't know why this is happening , and obviously neither do either of you unless of course I send the LOG files because you both work for Ultimaker but aren't privy to my name... makes no sense at all!

Ghostkeeper, 2021-08-09:
This discussion is quite off-topic as to the issue at hand. We're here to diagnose issues with Cura, not with Ultimaker. I'll hide it as such.

Cura is not supposed to update the Digital Factory or vice-versa. Both are updated by the printer firmware. Why this printer firmware seems to be able to update the Digital Factory, but not Cura, I don't know. This is what we hope to discover if you fill in the issue template with the (relevant) missing entry: the log file.

If you're unable or unwilling to provide the log file, we'll close this bug report as being incomplete automatically after 2 weeks.

TRLBY, 2021-08-09:
But, but, but you told me you worked for Ultimaker.... and now you're wearing the CURA hat. Doesn't Ultimaker Firmware folks beta test their updates with CURA, or do dump this on the users to sort out.

Hurry up now and hide this on this "public" forum.... we don't want anyone to know just frustrating this can be to the Ultimaker user.

But thanks for admitting that CURA and Ultimaker do not talk,  in fact it appears you don't even talk to yourself !

ayuhu, 2021-08-09:
Because I left a comment at the top I got all the notifications about the discussion... oh my...
While at first I understand some points about the "reporting-bug-situation" it seems that the discussion somehow got more important than the problem itself.

Now about the problem: I have it as well because I monitor all my printer through the Digital Factory on a Screen which runs 24/7. Therefore, all my printers are in the cloud. Since I don't print over my network anymore and only use the cloud or Library function for that I know for sure the printers are in the cloud.
Printing and monitoring works like it used to be (like camera live feed, Interactions, etc.), execpt the DF website isn't refreshing after 2 mins or so. I manually have to refresh the website to see the current status of the printers. After that I get like 1 or 2 automatic website refreshes but then it stops. 
I don't know for sure since I am not a part of Ultimaker but I would assume its an issue with the DF rather than the cloud connection of the printers. Therefore, I am not sure if a printer log file would help. 

I normally report things like this in the Ultimaker forum since I just like the forum more than github. Not that I know for sure, but I think posts like this, seen by moderators, can be directed faster to the right department. Just my personal feeling. 

My post in the forum already got acknowledged my @ChrisTerBeke 

Hope that helps...

Ghostkeeper, 2021-08-09:
Ah, but you have a different issue then, Ayuhu. The issue you're seeing is that the Digital Factory doesn't update, while Trlby was finding that Cura doesn't update. Indeed, with the Digital Factory not updating, a Cura log would be rather useless since the Digital Factory runs without Cura needing to be involved at all.
I'm supposing that was this issue: https://community.ultimaker.com/topic/37708-df-website-is-not-refreshing-anymore/

ayuhu, 2021-08-09:
This is correct. 
But I don't think that it is a coincidence that both errors appear at the same time. My error-searching-experience tells me that this is more likely one error somewhere in the system, rather than two in different locations. Besides that, the second picture from Trlby shows the DF as well. That's why I thought it is the same issue. 

Is the monitoring page from cura getting its information maybe from the DF or cloud? That could have somehow the same source then.

TRLBY, 2021-08-10:
My comments are now considered disruptive eh!   And all i did was expose the lack of communication between the CURA and Ultimaker.

And this is how a 3rd party forum for Software support works! If you don't like the criticism, then shut people out.

nallath, 2021-08-10:
> My comments are now considered disruptive eh! And all i did was expose the lack of communication between the CURA and Ultimaker.
> 
> And this is how a 3rd party forum for Software support works! If you don't like the criticism, then shut people out.

We didn't remove the comments, they are still there. This is a place to provide information on how to fix issues and request features. Since the discussion about internal Ultimaker policy is neither, they have been hidden to ensure that the thread remains readable. 

We still need log files to debug them, so please provide them so we can continue to find out what the issue is.

TRLBY, 2021-08-10:
Well I disagree with it being OFF SUBJECT,and I'd like to know who gets to decide this! I'd like someone from GitHub to contact me to answer these questions.

And how can you say: "his is a place to provide information on how to fix issues and request features. "  If neither you or Ghostkeeper don't want toanswer any of my questions? You deflect with requests for my LOG files, by claiming you're both Ultimaker Employees, yet neither of you offer NO PROOF as to who you are... even after telling me you have NO access to my LOG files that I've already posted to Ultimaker.  Why would anyone identified with non-de script  user names like Ghostkeeper and nallah be trusted with any files of mine? Is that NOT a fair question regarding the error I p[posted? 

I don't believe that's OFF TOPIC at all, but me simply asking for customer support, which by the way the CURA folks and Ultimaker sent me here! to get my problem solved. Yet somehow I'm expected some person named Ghostkeeper.. what a joke!

I'm sorry you don't like my "snarky" criticism of this silly idea of customer support, but maybe you 2 need to answer me how I'm suppose to know you're not both amateur moderators? Furthermore as "ayuhu" points out the LOG files won't help you at all since we don't know if this is a Firmware, CURA or Digital Factory problem.

Why hasn't "fmvr" responded to any of this?

nallath, 2021-08-10:
Okay, fine. Is this enough proof? 

Screenshot from my profile page (https://github.com/nallath) [image]

Screenshot from my membership flair: 
[image]

Screenshot from my linkedin (https://www.linkedin.com/in/jaimevankessel/): 
[image]

Do you also want a copy anything else while we're at it? Bank records showing that Ultimaker pays me perhaps? I only have signed employee contracts in Dutch, so I'm not quite sure if they would be helpful to you. 

> Furthermore as "ayuhu" points out the LOG files won't help you at all since we don't know if this is a Firmware, CURA or Digital Factory problem.

Uh. Yes it would. Since if there is a clear error in the logs, we would know more what is going on. 

This is the last time i'm even going to bother replying. Give the logs and get help, or don't give them (and don't get help). The choice is yours.

TRLBY, 2021-08-10:
My, my, my looks who's getting "snarky" now! Was that so hard to do that you couldn't do that 20 hours ago? So email me directly outside of Github and I'll send you the file.

Oh and just curious does Ghostyboy work with you?

nallath, 2021-08-10:
> My, my, my looks who's getting "snarky" now! Was that so hard to do that you couldn't do that 20 hours ago? 

I assumed that it was abundantly clear from my profile and flair that both of us are Ultimaker employees. So yes, the snark is there, since it doesn't matter if i'm actually an Ultimaker employee or a community member. There is _absolutely_ nothing sensitive in those log files and this discussion has wasted a lot of time, during which you could have already had a solution. We're here to try and help, but we need certain things to be able to actually do that. 

> So email me directly outside of Github and I'll send you the file.

I don't have your email address. We've already had this discussion. If you want you can send the logs to j.vankessel at ultimaker.com

> Oh and just curious does Ghostyboy work with you?

Yes. This is the last warning about calling people names. If you are unable to communicate without calling people names, I won't communicate with you at all. So lets keep it professional, shall we?

Ghostkeeper, 2021-08-10:
> Is the monitoring page from cura getting its information maybe from the DF or cloud? That could have somehow the same source then.

No, if the printer is connected to Cura via LAN, Cura requests that information directly from the printer with a HTTP get request. The printer has a mini-webserver that responds to HTTP requests with API information. You can see that same information in a browser too, if you know the printer's IP address. Just visit that IP address in a browser and it'll have a page for the API documentation with some examples that can be executed. The printer responds to API requests with a JSON file, that Cura parses and updates the monitor page with.

If the printer is connected to Cura via the cloud, Cura has more or less the same workflow but makes the requests to the Ultimaker website rather than directly from the printer. The API is slightly different due to authentication and feature parity but it works the same way from Cura's point of view.

ayuhu, 2021-08-10:
Was worth a shot... But thanks for the explanation.
Because TRLBY did also post a picture of his printer being in the DF I thought his printer also has to be in the cloud. That's why I thought cura might get the printer information from there as well. 

I just checked my monitoring page in Cura and left it open for a while. The print status with the progress bar updates while the print is going. Although I almost never use the monitoring page in cura since I have a screen running the DF 24/7 (which normally updates, hehe) it seems to be working for me. 

So it might be a network problem (or cura problem) for TRLBY. 
The interest to find the solution, however...

closed 2021-09-01 by no-response[bot]