Ultimaker/Cura #10217: 4.8.1 - Cura using T1 and T1 settings even when T1 is disabled.

tlhintoq, 2021-08-01:
### Application Version

4.8.1

### Platform

windows 10 x64

### Printer

Hictop D3Hero/Tenlog DL3pro

### Reproduction steps

With Extruder two (T1) disabled, slice.

The gCode is using T1 and all of T1 settings instead of T0.
Note that Cura added the T1 on line 22 BEFORE the custom start code.
And that the temp it picked was from T1, not T0 - same for all the other properties

[image]

### Actual results

Useing the disabled T1 instead of T0

### Expected results

Use the correct extruder

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fvrmr, 2021-08-02:
Hi @tlhintoq can you share your project file (file - save project) and gcode where this is happening? In that way it is easier to reproduce your issue.

Ghostkeeper, 2021-08-09:
Both Cura front-end and CuraEngine back-end are seeing that the initial extruder is T1, as evident by `T1` being put there by CuraEngine and the settings being evaluated for the initial extruder by the front-end. The initial extruder is [derived from the settings](https://github.com/Ultimaker/Cura/blob/176077c910b9dabe76f5c2047e9b87759f4cef0d/cura/Settings/ExtruderManager.py#L277-L296). It'll use the adhesion extruder, the support extruder or the extruder of one of the models.

The adhesion extruder and support extruder should get set to (one of) the remaining extruder(s) when they are set to be printed with the extruder you disable. However it is a known problem that if you have a profile that overrides the extruder selection setting, reverting that setting change is still possible, and also discarding changes when going to that profile still causes the disabled extruder to be selected. Perhaps this is what happened for you.

The current reproduction steps don't reproduce the issue for me with my own printer (Ultimaker 3). The reproduction steps can't be properly followed though, because Cura doesn't have any Hictop or Tenlog printer profiles, and this issue most likely depends on profiles.

tlhintoq, 2021-08-09:
funny - it depends on the Hictop profiles.  The same ones used for the last 4 years and how many versions of Cura - but just now started happening when the latest Cura was released.  Huh.

I'll see if I can reproduce it.  Honestly, I don't even remember which model I was slicing at the time.  I"ll dig into it.