Ultimaker/Cura #10218: Firmware Retract - too many extra retractions generated

stabbs, 2021-08-01:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Home brew

### Reproduction steps

When printing with multiple extruders Cura is adding unnecessary retractions in 4 places.  1) an initial G10 at the beginning of the G-code,  2) and 3) are reoccurring as they happen before and after each tool change, 4) is the last retraction in the g-code (not shown in the code snippet below)

I wanted to generate simple gcode using firmware retraction to illustrate this.  I place 4 20mm boxes in the plater, set them to 1 perimeter, 0 infill, and 0 top and bottom layers and sliced.  For simplicity assume retraction is 1mm and 'Nozzle Switch Retraction Distance' (referred to from now on as PARKED)  is 10mm:

*******  Starting with Extruder 0 (T0)
T0								
M82 ;absolute extrusion mode
G28 ;Home							
M190 S70 ; set bed temperature and wait for it to be reached
;G1 Z15.0 F6000 ;Move the platform down 15mm

*******  Cura assumes all prints will start with a prime
;Prime the extruder					
G92 E0
;G1 F200 E3
G92 E0
G92 E0
G92 E0

*******  First G10 generated by Cura. If this G10 could be added manually using 'Start-Gcode' it would give flexibility to those who want to prime the extruder and those who do not
G10								
;LAYER_COUNT:98
;LAYER:0
M107
;MESH:20mm-box.stl
G0 F3600 X94.3 Y99.8 Z0.3
;TYPE:WALL-OUTER

*******  Un-Retract - This should be the first retraction generated by Cura.  G11 will prime the nozzle from PARKED or retracted position
G11								
G1 F1800 X74.7 Y99.8 E0.36869
G1 X74.7 Y80.2 E0.73737
G1 X94.3 Y80.2 E1.10606
G1 X94.3 Y99.8 E1.47475
G0 F3600 X94.1 Y99.8
G0 X93.9 Y99.4						

*******  G10 Retract generated by Cura.  This should be a G10 S1 or omitted.  If kept as G10 this creates a big problem because we still need to park the extruder using a G10 S1. If kept as-is the extruder will G10 retract (-1mm) then G10 S1 (-10mm). T0 is now retracted 11mm when the true PARKED position should be -10
G10								
G1 F600 Z1.3

*******  The next line G10 S1 was added to 'Extruder End G-Code' in Cura to PARK the filament in the current extruder
G10 S1 ; begin tool change							
G92 E0
T1								
G92 E0

*******  The following G10 is generated by Cura.  A tool change has just occurred, why would we want to retract here? It should be assumed the filament is already PARKED or retracted.  This G10 should be omitted
G10								

;MESH:20mm-box.stl(3)					
G0 F3600 X93.9 Y99.4 Z1.3
G1 F600 Z0.3
G0 F3600 X95.72 Y99.4
G0 X115.2 Y99.78
G0 X115.3 Y99.85
;TYPE:WALL-OUTER					

*******  No E moves up to this point since last tool change. The following G11 should be kept here because it will un-retract from both PARKED or retracted positions
G11								
G1 F1800 X95.7 Y99.85 E0.36869
G1 X95.7 Y80.25 E0.73737
G1 X115.3 Y80.25 E1.10606.......

******** There is a final G10 move at the end of the g-code that should be omitted when using multiple extruders.  The final retraction should be a G10 S1 to PARK the filament.

### Actual results

Without manually editing the G-code the extruder adds unnecessary retractions. Most prints fail quickly because the filament is over retracted.  

### Expected results

Simplified requested code without comments:

T0								
M82 ;absolute extrusion mode
G28 ;Home							
M190 S70 ; set bed temperature and wait for it to be reached
;G1 Z15.0 F6000 ;Move the platform down 15mm
G92 E0
						
;LAYER_COUNT:98
;LAYER:0
M107
;MESH:20mm-box.stl
G0 F3600 X94.3 Y99.8 Z0.3
;TYPE:WALL-OUTER
G11								
G1 F1800 X74.7 Y99.8 E0.36869
G1 X74.7 Y80.2 E0.73737
G1 X94.3 Y80.2 E1.10606
G1 X94.3 Y99.8 E1.47475
G0 F3600 X94.1 Y99.8
G0 X93.9 Y99.4						
							
G1 F600 Z1.3
G10 S1 ; begin tool change							
G92 E0
T1								
G92 E0							

;MESH:20mm-box.stl(3)					
G0 F3600 X93.9 Y99.4 Z1.3
G1 F600 Z0.3
G0 F3600 X95.72 Y99.4
G0 X115.2 Y99.78
G0 X115.3 Y99.85
;TYPE:WALL-OUTER					
G11								
G1 F1800 X95.7 Y99.85 E0.36869
G1 X95.7 Y80.25 E0.73737
G1 X115.3 Y80.25 E1.10606

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I've read through Cura's source code but I know very little about programming and couldn't figure out a solution on my own.  Perhaps the retraction logic makes sense on some 3d printers but in my case its creating problems that can only be fixed by me manually editing the g-code.

fvrmr, 2021-08-02:
Hi @stabbs could you share a project file (file - save project) in that way it is easier to reproduce your issue.

stabbs, 2021-08-02:
Sure, here it is
[CFFFP_20mm-box.zip](https://github.com/Ultimaker/Cura/files/6918321/CFFFP_20mm-box.zip)

Ghostkeeper, 2021-08-09:
A bit unrelated, but slicing that project file fails when CuraEngine is compiled in debug mode, with the following assertion failure:

    void cura::LayerPlanBuffer::insertFinalPrintTempCommand(std::vector<cura::ExtruderPlan*>&, unsigned int): Assertion `(time_window >= 0 || last_extruder_plan.estimates.getMaterial() == 0) && "Time window should always be positive if we actually extrude"' failed.

So something does seem to go wrong there, perhaps with the time estimations.

> First G10 generated by Cura. If this G10 could be added manually using 'Start-Gcode' it would give flexibility to those who want to prime the extruder and those who do not

Cura assumes that the filament starts at the nozzle tip, after the start g-code. If people don't want to prime the extruder, they will need to at least make sure that the filament is in the nozzle tip so that it is ready for printing if Cura decides to print immediately. The start g-code cannot make assumptions based on what Cura is going to do afterwards, because it may change due to settings and the starting conditions of the print. This G10 can be removed by disabling retractions as a whole (but of course you don't want to do that). There is no separate control for this individual retracted move going from the start position to the print. And I don't think there should be.

In other words, giving that flexibility would require some form of communication for Cura to know what position the filament ended up in after the start g-code. This communication is difficult for the user to configure, so we just assert that the position needs to be at the nozzle tip.

> G10 Retract generated by Cura. This should be a G10 S1 or omitted. If kept as G10 this creates a big problem because we still need to park the extruder using a G10 S1. If kept as-is the extruder [...]

I believe this is where your real problem starts. Cura doesn't support the `S` parameter of the `G10` command at all. So the nozzle switch retract is identical to the normal retract. You then add a nozzle switch retract yourself with the extruder switch g-code settings. With software-retractions there is a difference in the amount of E that the material retracts, but with firmware-retractions there is no such difference any more.

Since this is just a feature that is not implemented by Cura, I'll change this to a feature request.

> The following G10 is generated by Cura. A tool change has just occurred, why would we want to retract here? It should be assumed the filament is already PARKED or retracted. This G10 should be omitted

Again, Cura will assume that all extruders start with the filament at the nozzle tip.

stabbs, 2021-08-10:
Thank you for the explanation Ghostkeeper and for updating my question to a feature request.

> Again, Cura will assume that all extruders start with the filament at the nozzle tip.

I respectively disagree with Cura's logic here. Cura should assume the extruder(s) are in their retracted positions at the start of the g-code  Having idle extruders primed results in unnecessary ooze.  I understand this can be mitigated to a degree with temperature control but I find retraction to be a better solution.  

The final E move generated in cura's g-code is a retract.  Any follow-up print may fail because Cura assumes the filament is at the nozzle tip but is in fact retracted.  Priming at the start of the g-code will remedy this if the prime distance > retraction distance but priming every print should not be a required procedure.  I suggest at the start of a print on single extruder printers Cura should assume filament is retracted.  For >1 extruder setups. all extruders filament should be assumed in the 'G10 S1' swap position or else retracted.

3Duke3, 2021-08-13:
I agree that there is an oddity here. While the original issue description seems rather complex, I think its very relevant to prime related issues I experience on both my toolchanger and IDEX printers, that I have not managed to work my way around yet, and would be super keen to hear what the solution is, rather than using another slicer program.

I would like to perform some sort of filament purge at every tool change by writing some gcode in the "Extrudes Start G-code" fields, or using a macro on the Duet. The problem being on both printers, is that when a tool is picked up for the first time, Cura expects filament at the tip of the nozzle, however when it picks up the tool every subsequent time, Cura then expects the filament to be at -{retraction_amount} position, or -{switch_extruder_retraction_amount} position, if a "nozzle switch Retraction Distance" is set.

I'd like to have both 'retraction' and, 'nozzle switch retraction' handled by Cura's gcode as it makes a much cleaner retract off the printed part when changing between tools, but as the 'extruder start g-code' is embedded for the first, and every following tool change, its impossible to position the filament correctly for both the initial and subsequent uses. 

For my 'extruder start g-code' I have a small list of gcode to purge some filament, wait a second, and then reposition the filament back by the nozzle switch retraction amount of 2mm. This is to accommodate every subsequent toolchange after the first and it works great. However it does not work for picking up a tool for its first use, because we have moved the filament back 2mm, and Cura expects it to be ready to go at the nozzle tip, resulting in underextrusion at the beginning of every tools first use. It's not so bad when both/all tools are used at the beginning of the print, but when a new tool is activated 4h into a print, I'm not always ready to watch and assist it.

I don't quite know what the solution is here, but I also agree that the logic is not best practice. I am really hoping there is a simple solution to this.

Ghostkeeper, 2021-08-18:
Did you try the Extruders Share Nozzle machine setting? It's not precisely what your printer actually has, but it changes the priming logic to allow for there being only one filament in the nozzle at a time. After all, if there is only one nozzle on the printer, there can only be a single filament in that nozzle. Maybe this is a solution for your printers.

Ultimaker's printers have separate nozzles (not IDEX since they are on a single head, but still). These printers handle this issue in firmware.

The problem with assuming that the filament is retracted at the start, is that we have to know for every single printer on the market how far it is retracted, and change the starting routine for all of them. We don't know this, and we can't change this logic until we know this for every printer we have a profile for, and also for all printers people might have made custom profiles for. This is obviously impossible to know, so we can't change this logic, ever, by default.

3Duke3, 2021-08-18:
Thanks @Ghostkeeper. I actually did some more tinkering using "Firmware Retractions" on my toolchanger, and have it working reasonably well. The only downside is at the end of a tools movement, the filament is retracted and the Z axis raised which pulls a small stand/string of filament directly upward, this does not appear to happen when using Cura retractions, but I am unsure if its because the retraction + z hop is much more seamless, or if Cura manages to insert a tiny wipe as it performs this move. I could of course remove the z hop on extruder switch, but this setting also has its benefits.

Looking through the generated gcode, if "firmware retraction" is enabled, its assumed the filament is starting at the tip of the nozzle each time a new tool is brought out, this is despite each tool finishing with a retraction before its put away. I don't quite understand why the filament position is remembered for Cura retractions, but not for firmware retractions. However this actually works favorably as I can run a prime script when the tool is brought out, and it seamlessly primes (from the script), retracts (Cura G10), moves to position for printing, then un-retracts (Cura G11), and were away printing.

Interesting you mention the extruders share nozzle setting, it never crossed my mind to look at this, but makes sense now you mention it. I am going to have a play with this tomorrow as it sounds like it could be an ideal option for my (marlin) IDEX printer.

Ghostkeeper, 2021-08-18:
> but I am unsure if its because the retraction + z hop is much more seamless, or if Cura manages to insert a tiny wipe as it performs this move. I could of course remove the z hop on extruder switch, but this setting also has its benefits.

In CuraEngine, the retraction command is one-to-one replaced with `G10` if firmware retractions are used. It does indeed wipe before retracting if the nozzle is at that point on the outer wall. It should wipe just the same with firmware retractions though.

During a tool switch later in the print, Cura does not assume that the filament remains at the nozzle tip (that's only at the start of the print). It should be retracted by the Nozzle Switch Retraction Distance, and behave that way. When using firmware retractions, this Nozzle Switch Retraction Distance is remembered internally but in the g-code ends up as simply `G10`, and `G11` to unretract. This is what I believe to be the original issue of not supporting the `S` parameter.

rubienr, 2023-09-10:
I suffer from same issue, namely that on tool change Cura sends retract `G10` instead of swap retract `G10 S1`. Since Marlin allows `G11` only after `G10` and `G11 S1` only after `G10 S1` my workaround is:

Start G-Code (extruder 1 and extruder 2)
``` g-code
G11 S1 ; swap-retract recover
```
End G-Code (extruder 1 and extruder 2)
``` g-code
G11    ; recover: undo last G10 retract  (Cura bug workaround)
G10 S1 ; swap-retract
```
This removes oozing completely. However, it is a quick and dirty workaround and the produced G-Code contains unnecessary movements (retract + undo retract). IMHO Cura shall generate `G10 S1` and `G11 S1` on tool change respectively.