Ultimaker/Cura #10219: Older version error

Sur04, 2021-08-02:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

A message pop-ups in ultimaker S5 saying " This file is made with an older version of Ultimaker Cura. For optimal performance prepare your print with Ultimaker Cura version 4.4 or higher." While slicing in latest version of Cura(4.10.0).

### Actual results

Such pop-up shouldn't appear.

### Expected results

No message.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It is actual version bug because such message doesn't appear in other versions.

fieldOfView, 2021-08-02:
This has been fixed in newer firmware available for your printer.

fvrmr, 2021-08-02:
Hi @Sur04 this is indeed fixed in the firmware version 6.4.0
Also see https://github.com/Ultimaker/Cura/issues/10062 
I will close this issue as a duplicate. But please reach out if you have more questions.

Sur04, 2021-08-02:
Thank you for the prompt solution.