Ultimaker/Cura #10220: CURA 4.10 does not import profiles or printers from 4.9.1

Xoxe2, 2021-08-02:
### Application Version

4.10

### Platform

OS X 10.13.6 High Sierra

### Printer

Personalized Kingroon KP3S

### Reproduction steps

The update to CURA 4.10 does not update the printer or the custom profiles that I have created for each printing assumption, resolution, speeds, retraction, flow, and many more details created in 9.0 and updated to 9.1. Creating a new printer in 4.10 gives me an error when importing these profiles.

### Actual results

[image]

[image]

Error al importar el perfil de /Users/xoxeluiz/Documents/perfiles cura/KP3S Low Quality FLOW.curaprofile: El tipo de calidad 'low' no es compatible con la definición actual de máquina activa 'fdmprinter'.

### Expected results

Import custom files 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-08-02:
Hi @Xoxe2 this has to do with that your printer doesn't have the low quality profile. And the profile you want to import is based on that profile.
You can find here more information about importing profiles: #8229 
Did you create these profiles yourself in 4.9?

Xoxe2, 2021-08-02:
Yes. Create a new Kongroon KP3S printer indicating the parameters and the start and end g-codes.
[image]
[image]

Xoxe2, 2021-08-03:
I have solved the CURA 4.10 problem by creating a printer profile with the "Creality Ender 2" associated with the "nozzle" and modifying its code to suit the KP3S specifications and changing the name of the printer to my liking. I have installed a custom dock for the KP3S in the "meshes" folder.
All this within the CURA application, in the "resources" folder and each file in the corresponding folder. OS X (Mac)
I have inserted the starting and ending g-codes directly into CURA.
I have been able to recover all my old custom files,

fvrmr, 2021-08-03:
Hi @Xoxe2 great to hear that you've found a way to make it work!

closed 2021-08-05 by fvrmr