Ultimaker/Cura #10223: Add material used to Gcode for e.g. Filament Manager Plugins in OctoPrint

NilsRo, 2021-08-03:
### Is your feature request related to a problem?

My filament runs out.  :-)

### Describe the solution you'd like

Add material used (including all important information like color,... ) to GCode. Actually it is missing in the Start-/Endcode variables: http://files.fieldofview.com/cura/Replacement_Patterns.html

material GUID is available but useless as it is the same GUID for custom materials.

### Describe alternatives you've considered

Material has to be selected manually in the Plugin before every print (if OctoPrint is used)

### Affected users and/or printers

All printers if OctoPrint is used, also helpful to check for which material the GCode files is created.

fieldOfView, 2021-08-03:
> material GUID is available but useless as it is the same GUID for custom materials.

The material GUID is only the same if you keep duplicating materials to create a new one. If you üse the "Create" button instead of "Duplicate", you get a new GUID for each newly created material. For materials that you "Duplicate"d before, there's an "Unlink" button somewhat hidden below the material properties:
[image]

NilsRo, 2021-08-03:
I like to get it linked as for example I created 4 new materials out of PETG like to adapt some properties only. Or Made a brand PETG material and like to duplicate it for every color I own. What you describe is a workaround as the management of the materials is higher as it does not share basic properties (as I understood the behaviour in the correct way).

So If I duplicate green PETG, yellow PETG, ... from generic PETG I like to change the print temperature for generic PETG only.

Ghostkeeper, 2021-08-16:
The GUID is meant to indicate a type of spool. For instance, all PLA Red spools by Ultimaker have the same GUID, but the PLA Green spools by Ultimaker have a different GUID. The "shared properties" are also the ones pertaining to types of spools, e.g. filament cost and weight. You can still have multiple profiles for the same spool, e.g. to print PLA Red with different temperatures or fan speeds.

It does not indicate which individual spool is being used. Cura has no information on that. It would be rather cumbersome to have to tell Cura that the spool got changed, something which any material manager would already know anyway.

But yeah, the GUID is the intended solution here. The Ultimaker firmware uses this GUID to ask the user to change filament if a different type of spool is currently in the printer, for instance. OctoPrint can do this just as well, but to get other metadata like filament colour or the name of the profile, it would have to get a copy of the fdm_materials repository to match those GUIDs with the built-in profiles, and it would have to receive the profile set over the network, whenever the users create their own material profiles.
We've implemented all of that for Ultimaker printers. Cura sends over those material profiles and the printer firmware receives them and parses them and matches the GUIDs in the g-code files with the materials it has in its own database so that it can display all sorts of metadata about the material, and also so that it can heat to the correct temperatures while calibrating etc. We are not about to implement that for Octoprint.

Alternatively to the GUID, Cura will also put the profile ID in comments at the bottom of the file. These profile IDs are human-readable, but can't be controlled by the user so if they duplicated a material it'll get a name like `generic_pla_%232`, encoded from `Generic PLA #2`.

So to discuss the three use cases you mentioned:

* > manually check for which material the GCode was created
  
  This can be done by reading the profile ID in the bottom.
* > show material in OctoPrint Filebrowser (more comfort)
  
  This is the difficult case where OctoPrint would have to have a copy of the material profile so that it can look up all of the information it needs, like material colour and profile name.
* > make material identifiable for a filament manager to monitor the filament left on a role
  
  For this you'd need to know the individual spool rather than the spool type. Otherwise swapping from Red PLA to Red PLA would be indistinguishable. Cura is not able to provide a solution for this.

For the second case, the implementation would currently be quite involved. We only provide the GUID right now, because the firmware is going to need a shadow copy of the entire profile anyway for Ultimaker's requirements. After all, the Ultimaker printers make calibration prints which need basically all printing parameters anyway. If OctoPrint doesn't need all of that information but only selected parts of the material profile, then it would make the implementation simpler on OctoPrint's side if Cura can provide metadata about the currently active material profiles for every extruder in the g-code with extra replacement keys, similar to `{time}` or something, which already exists as a replacement key.

However since this is only applicable to third party software integration, Ultimaker won't build this for you. We will defer this until either Ultimaker needs this for one of their applications, or until someone wishes to spend their own time on implementing this for OctoPrint.

fieldOfView, 2021-08-16:
> We will defer this until (...) someone wishes to spend their own time on implementing this for OctoPrint.

See #10289

NilsRo, 2021-08-16:
> For instance, all PLA Red spools by Ultimaker have the same GUID, but the PLA Green spools by Ultimaker have a different GUID. 

@Ghostkeeper : That sounds like the perfect behaviour but I cannot reproduce it for custom materials. A new GUID is created using "Create"-Button like @fieldOfView suggested only. If I duplicate a material to copy also the "Print settings" (as it is the same color it would be very easy to create the new material) the GUID is not changed. Perhaps it is a bug is it does not looks like a good behaviour in general. Or is this you described with:
> You can still have multiple profiles for the same spool, e.g. to print PLA Red with different temperatures or fan speeds. 

Duplicate --> GUID is duplicated also (is it a GUID then --> two materials, different color, same GUID?)
Create --> new GUID is created
Unlink --> GUID is not changed also (but that is perhaps fine, as I know it is for the properties only). 

[image]

PS. I also now understand that only properties are shared and not print settings. There I was wrong before...
PPS. I still developed an OctoPrint Plugin so enhancing it would not be a problem. I can start with the enhanced replacement patterns (@fieldOfView many thanks) but like to understand the GUID also, perhaps it is better at the end... (e.g. uploading the material files from Cura to OctoPrint to take over the material settings...)

Ghostkeeper, 2021-08-23:
> Unlink --> GUID is not changed also (but that is perhaps fine, as I know it is for the properties only).

This is unexpected. The unlink should change the GUID. That's more or less the whole point of unlinking materials.

NilsRo, 2021-08-23:
@Ghostkeeper Ok, then this is a bug that confused me. There are some forum-discussion I found that custom materials have the same GUID everytime. I think this sentence is based on that bug-behavior. I tried it with Cura 4.10.0 and can easily reproduce it. Materials duplicated shares the same GUID forever.

Also I cannot remove the unlinked material anymore even if another material is active.

[image]

fieldOfView, 2021-08-23:
> I tried it with Cura 4.10.0 and can easily reproduce it. Materials duplicated shares the same GUID forever.

Yes, that is what duplicating means. The duplicated material is the same (a duplicate), so it has the same ("duplicated") GUID. Materials that are created with the "Create" button always have a unique GUID. The unlink button is specifically there to create a new GUID so duplicated materials are no longer duplicates.

NilsRo, 2021-08-23:
Should I create a new Issue that Unlink is not working fine?

Ghostkeeper, 2021-09-02:
You can, as long as that issue is not a duplicate of https://github.com/Ultimaker/Cura/issues/10235, which we fixed for 4.11. Did you try unlinking the materials in 4.11?

This issue is tracking the feature to allow OctoPrint to see the material name and type. It'll be closed once https://github.com/Ultimaker/Cura/pull/10289 is merged.

nallath, 2021-09-24:
So as per instructions i will close this.

closed 2021-09-24 by nallath

NilsRo, 2021-09-26:
> 
> 
> You can, as long as that issue is not a duplicate of #10235, which we fixed for 4.11. Did you try unlinking the materials in the 4.11 beta?
> 
> This issue is tracking the feature to allow Octoprint to see the material name and type. It'll be closed once #10289 is merged.

@Ghostkeeper : Created incident https://github.com/Ultimaker/Cura/issues/10516 as it happens also in 4.11