Ultimaker/Cura #10225: No handshake w/Anycubic Mega Zero 2.0

eltoneo, 2021-08-03:
### Application Version

4.10.0

### Platform

macOS Big Sur

### Printer

Anycubic Mega Zero

### Reproduction steps

Reinstalled Ultimake_Cura-4.10.0-Darwin.dwg.

### Actual results

No change
ATT: Cura help
Referenced app slices ok, however 'Jog Position' X/Y and Z are all dead under Monitor!
'Print via USB' displays with Preview and '/dev/cu.usbserial-1420' displays under Monitor.

There is no sign that a handshake is made to the Anycubic Mega Zero 2.0 printer.
Also Cura Log displays an ERROR, and a detected baud rate of 250,000.

### Expected results

Jog Positions X/Y and Z should be activated on printer, but is dead.
Also, no sign the gcode reaches printer!

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6924984/cura.log)

There is no sign that a handshake is made to the Anycubic Mega Zero 2.0 printer.
Also Cura Log displays ERROR[S], and a detected baud rate of 250,000.
Please help resolve this Cura bug.
eltoneo
* The Mega Zero 2.0 requires a baud rate of 115200 *

eltoneo, 2021-08-04:
This thread refers to late 2013 Macbook Pro w/Intel 64-bit CPU

fvrmr, 2021-08-06:
Hi @eltoneo thank you for your bug report. But we don't test this with your printer, we only test with Ultimaker printers. So I can't really help you with that. I hope you understand.

closed 2021-08-06 by fvrmr