Ultimaker/Cura #10227: Support does not show when other user opens project file

LilBub, 2021-08-03:
### Application Version

4.9 Enterprise

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1.  Open project file
2.  View print settings and preview

### Actual results

One person can see the supports used in the print
If another person opens the very same project file, the supports are not shown

### Expected results

50/50 chance of seeing supports 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Sample Cura Projects.zip](https://github.com/Ultimaker/Cura/files/6925742/Sample.Cura.Projects.zip)

GregValiant, 2021-08-04:
I don't know if this option is supposed to be saved within a 3mf file but it appears that it is not.  It doesn't have anything to do with the model or slicing settings.  It appears that each user must enable or disable it on their own computer.
[image]

LilBub, 2021-08-05:
That's an interesting catch.  
One of my staff noticed that as well this morning

Will future versions address this?

Ghostkeeper, 2021-08-16:
I wouldn't consider this a bug. Which structures you want to see is part of the user preferences, not part of the print. It's intended to work that way.

To explain why, anecdotally I like to only show travel moves and starts when I'm debugging something related to those. I'd find it pretty annoying that it would override my preferences when I'm loading in someone else's project file, if that project file was made by someone who prefers to always have travel moves visible.

I'd consider the fact that this is loaded from a project file to be unrelated to the actual issue. Had the user just loaded an STL that needed support, and enabled support in the settings, they'd have had the same problem. The actual issue is that the user had this checkbox disabled, and forgot about it or didn't realise that it would apply to support too. This is a user experience issue.

LilBub, 2021-08-16:
When engineers send my team prints, they send the project file, since only the STL would not include their particular settings (i.e. Support).  When the Project file is sent, we can view everything they selected for options.  

So, as in this issue, sometimes we could see they had support material selected, others could not.

If there is a better process for engineers to send their work to Print Operators, we are open to ideas.

To date, the Project Files have worked well.  

Due to our secure environment, we only print via USB and not the Cura queue system.

Ghostkeeper, 2021-08-16:
Using project files is fine of course. What I was saying is that this issue is not *unique* to using project files. It simply has very little to do with using project files or not.

LilBub, 2021-08-16:
Well, it would be an improvement opportunity for the UI to detect differences in the Legend display settings (Helpers off?)

Ghostkeeper, 2021-08-16:
That's indeed the direction in which I'm looking for solutions now. But I'm not really sure what you mean there.

What would it show the difference between? Between showing helpers and not showing helpers? It already shows the difference by there being a checkmark or not.

LilBub, 2021-08-16:
I'm trying to think of an example and here's the best one I can conjure up.
Such as when you change the Print Profile.  You get the pop up to Keep or Discard Changes prompt

Ghostkeeper, 2021-08-23:
I'm still not really understanding what you meant with that. You want it to show a pop-up? When should the pop-up be triggered, and what should it show?

There is already an interface that shows which structures are turned off. It's this panel:
[image]

Again, support being hidden has nothing to do with loading project files. It's not modified by project files. It's a bit like saying that you can't see support after taking off your shoes, so Cura should show a pop-up when you take off your shoes to notify that support might not be visible. Taking off your shoes has nothing to do with showing support or not. The pop-up would just be annoying then.