Ultimaker/Cura #10228: Print temperature +10°C after gcode export

mvsoldier, 2021-08-03:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Anycubic Mega X

### Reproduction steps

Set the Print Temperature to 200°C 
and exported it as a gcode

### Actual results

the Temperature in the Exported gcode is set to 210 °C

### Expected results

the print Temperature should be 200 °C

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I Tried a different Anycubic printer and the same export bug happend

fieldOfView, 2021-08-03:
Without a project file, we cannot see where Cura gets it temperature from in your case, so we cannot help you.

closed 2021-08-06 by mvsoldier

Ghostkeeper, 2021-08-16:
Most likely the difference between the normal printing temperature and the initial layer printing temperature.