Ultimaker/Cura #10230: Cura crashing when opening STL files

ivaanto, 2021-08-04:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Biqu B1

### Reproduction steps

The problem is only happening in version 4.10.0. In previous versions the program is working perfectly. Every time I try to open any STL file in Ultimaker Cura the program goes unresponsive and then abruptly closes.

### Actual results

The same problem is happening

### Expected results

that the program works

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

2021-08-04 10:52:34,837 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-08-04 10:52:34,875 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdateChecker
2021-08-04 10:52:34,900 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdater
2021-08-04 10:52:34,926 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzReader
2021-08-04 10:52:34,950 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzWriter
2021-08-04 10:52:34,976 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeProfileReader
2021-08-04 10:52:35,019 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeReader
2021-08-04 10:52:35,046 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeWriter
2021-08-04 10:52:35,081 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ImageReader
2021-08-04 10:52:35,112 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LegacyProfileReader
2021-08-04 10:52:35,131 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalContainerProvider
2021-08-04 10:52:35,158 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalFileOutputDevice
2021-08-04 10:52:35,179 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MachineSettingsAction
2021-08-04 10:52:35,209 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ModelChecker
2021-08-04 10:52:35,227 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MonitorStage
2021-08-04 10:52:35,271 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PerObjectSettingsTool
2021-08-04 10:52:35,301 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PostProcessingPlugin
2021-08-04 10:52:35,328 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PrepareStage
2021-08-04 10:52:35,351 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PreviewStage
2021-08-04 10:52:35,391 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RemovableDriveOutputDevice
2021-08-04 10:52:35,414 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SentryLogger
2021-08-04 10:52:35,479 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SimulationView
2021-08-04 10:52:35,507 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SliceInfoPlugin
2021-08-04 10:52:35,539 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SolidView
2021-08-04 10:52:35,569 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SupportEraser
2021-08-04 10:52:35,679 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-08-04 10:52:35,686 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-08-04 10:52:35,696 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-08-04 10:52:35,703 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-08-04 10:52:35,710 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-08-04 10:52:35,718 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-08-04 10:52:35,726 - WARNING - [MainThread] UM.Logger.logException [112]: Exception: Failed to load packages for Marketplace
2021-08-04 10:52:35,732 - WARNING - [MainThread] UM.Logger.logException [116]: NoneType: None
2021-08-04 10:52:35,749 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin Toolbox
2021-08-04 10:52:35,781 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CameraTool
2021-08-04 10:52:35,817 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MirrorTool
2021-08-04 10:52:35,872 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RotateTool
2021-08-04 10:52:35,911 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ScaleTool
2021-08-04 10:52:35,937 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SelectionTool
[... 695 lines cut ...]
2021-08-04 11:00:54,225 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 4.3s
2021-08-04 11:01:03,542 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.04700613021850586 seconds
2021-08-04 11:28:39,198 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.031168699264526367 seconds
2021-08-04 11:41:20,990 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1758]: Attempting to read file file:///C:/Users/Ivã Antonio/Downloads/Stunning Hango.stl
2021-08-04 11:41:21,012 - INFO - [JobQueueWorker [3]] STLReader.STLReader.load_file [53]: Using NumPy-STL to load STL data.
2021-08-04 11:41:21,044 - DEBUG - [JobQueueWorker [3]] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.0 seconds
2021-08-04 11:41:21,059 - DEBUG - [JobQueueWorker [3]] STLReader.STLReader._read [105]: Loaded a mesh with 1614 vertices
2021-08-04 11:41:21,290 - CRITICAL - [JobQueueWorker [3]] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2021-08-04 11:41:21,296 - CRITICAL - [JobQueueWorker [3]] cura.CrashHandler.__init__ [69]: AttributeError: 'NoneType' object has no attribute 'close'
2021-08-04 11:41:21,883 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [112]: Exception: Failed to open file b'C:\\Users\\Iv\xc3\xa3 Antonio\\AppData\\Local\\Temp\\scipy-hvolnwax'
2021-08-04 11:41:21,895 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]: Traceback (most recent call last):
2021-08-04 11:41:21,905 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshFileHandler.py", line 45, in readerRead
2021-08-04 11:41:21,915 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 220, in getExtents
2021-08-04 11:41:21,925 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 296, in getConvexHullVertices
2021-08-04 11:41:21,935 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 286, in getConvexHull
2021-08-04 11:41:21,941 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 276, in _computeConvexHull
2021-08-04 11:41:21,947 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 495, in approximateConvexHull
2021-08-04 11:41:21,952 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 512, in createConvexHull
2021-08-04 11:41:21,959 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "qhull.pyx", line 2433, in scipy.spatial.qhull.ConvexHull.__init__
2021-08-04 11:41:21,965 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "qhull.pyx", line 270, in scipy.spatial.qhull._Qhull.__init__
2021-08-04 11:41:21,971 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]:   File "messagestream.pyx", line 36, in scipy._lib.messagestream.MessageStream.__init__
2021-08-04 11:41:21,977 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]: OSError: Failed to open file b'C:\\Users\\Iv\xc3\xa3 Antonio\\AppData\\Local\\Temp\\scipy-hvolnwax'
2021-08-04 11:41:21,986 - WARNING - [JobQueueWorker [3]] UM.Mesh.MeshFileHandler.readerRead [62]: Unable to read file C:/Users/Ivã Antonio/Downloads/Stunning Hango.stl
2021-08-04 11:41:21,993 - DEBUG - [JobQueueWorker [3]] UM.FileHandler.ReadFileJob.run [75]: Loading file took 1.0 second
[cura.log](https://github.com/Ultimaker/Cura/files/6932111/cura.log)

nallath, 2021-08-04:
Have you checked this link already? https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

ivaanto, 2021-08-04:
Yes, Jaime. I've researched the problem before, and as for this log you sent, I've never had the Chitubox program installed on my machine.

 

Best Regards,

 

Ivã

Enviada em: quarta-feira, 4 de agosto de 2021 11:55
Para: Ultimaker/Cura ***@***.***>
Cc: ivaanto ***@***.***>; Author ***@***.***>

fvrmr, 2021-08-05:
Hi @ivaanto this is a duplicate of https://github.com/Ultimaker/Cura/issues/9644
For the next release we have a fix that Cura doesn't crash anymore.

Here is what I found in your log:
`2021-08-04 11:41:21,977 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]: OSError: Failed to open file b'C:\\Users\\Iv\xc3\xa3 Antonio\\AppData\\Local\\Temp\\scipy-hvolnwax'` 

The temp folder doesn't have access because of non-Latin characters in your Windows username. A workaround could be changing the path of the `%temp%` folder in the Windows environment variables configuration screen. 

I will close this issue as a duplicate.

closed 2021-08-05 by fvrmr