Ultimaker/Cura #10231: Z feedrate too high

Jarren141, 2021-08-04:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Custom FFF

### Reproduction steps

created custom Machine Def file for my printer 
added a override where max z feedrate is 1200
sliced file

### Actual results

File sliced correctly but had a very high feedrate for just this one line at the begining
**;LAYER_COUNT:80
;LAYER:0
M107
**G0 F31200 X272.775 Y188.954 Z0.25**
;TYPE:SKIRT
**

### Expected results

The feedrate for z would be either a seperate g0 line or just reducing this feedrate to the max feedrate for z

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Def.json files for this printer
[Cura Bug.zip](https://github.com/Ultimaker/Cura/files/6933469/Cura.Bug.zip)

fieldOfView, 2021-08-04:
In the frontend, and in definition files, feedrates are defined in mm/s. In gcode the feedrate is normally specified in mm/min. I think the number you entered in your custom definition is a factor 60 too big.

Jarren141, 2021-08-04:
I updated the definition value to "machine_max_feedrate_z":{"default_value": 50}, and restarted Cura but got the same result of 

G0 F3600 X329.597 Y239.046 Z0.3

 I even tried a value of .001 and got the same result.  // "machine_max_feedrate_z":{"default_value": 0.001}, // and got a result of 

G0 F3600 X329.597 Y239.046 Z0.3

GregValiant, 2021-08-05:
I don't understand the problem.  Those look like travel moves and so they are at your "Travel Speed" which appears to be 60mm/sec.  The max Z speed should not have been involved.
Max Z Speed is held within the printer in M203.  My printer's response to M503 contains:
echo: Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z50.00 E25.00
As far as I know Cura never interferes with M203 (although the author of a definition file or a user could put it in the Start Gcode).  The Max Z Speed in Cura's Printer Settings appears to only provide maximums for things like the Z Hop Speed textbox in the Speed settings and some speed textboxes for Ironing and Wireframe.  An example would be...

"speed_z_hop":  { "maximum_value": "machine_max_feedrate_z"}  from fdmprinter.def.json.

Without a maximum defined for "machine_max_feedrate_z" a user can put in anything they want.  The default is where they would start.

Ghostkeeper, 2021-08-16:
This is marked as "needs info". Can we specify what info is needed here?

I can reproduce the symptom with the given definition files, at least. Seems fine to me.

Ghostkeeper, 2021-08-16:
Cura assumes that the start g-code leaves the nozzle at position 0,0,0. But even without that assumption, the Z feedrate is really only [0.26mm/s](https://www.wolframalpha.com/input/?i=%281+-+0.3%29+%2F+sqrt%28%28321.329+-+334.597%29%5E2+%2B+%2862.96+-+224.046%29%5E2+%2B+%281+-+0.3%29%5E2%29+*+%283600%2F60%29). This is way below the specified maximum feedrate of 50mm/s. I wouldn't consider this a problem.

CuraEngine doesn't check the limits for this movement at all though. It is a bit of an edge case since it is the very first travel move, not really a "layer change" as such. It is not made as part of the layer change routine. This is why the limit is not held. It is mainly just considered to be an X/Y movement, although it also has a Z component. In your case the X/Y dimensions moved 160 times as much as the Z dimension, so it is very unlikely that the Z dimension is going to give a problem.

If the Z dimension is indeed limited even though the X/Y dimensions are moving much faster, the time estimation that Cura gives will be off. The firmware will still limit the Z speed, of course. Cura can't override that.

I consider it to be extremely rare that the Z dimension will be limited so much that the 0.3mm Z distance is a limiting factor in the initial travel move. And if it does occur, the time estimates will be off by about a second, which is going to be barely noticeable, especially considering that Cura's assumption for the starting position is also 400mm off in this case which will be MUCH more of an influence on the time estimate.

As such, I don't consider this issue to be worth our time to implement now. There is a theoretical bug there, but it's not in the way of anyone.

Putting the Z component on a different line will be more in the way for users than this issue.

Jarren-Essentium, 2023-05-26:
It seems like the behavior is still the same in Cura 5.3.0.

Here is the configuration set in the machine definition json

"machine_max_feedrate_z": { "value": 30 },

and here is a section of the output gcode
```
;LAYER:0
M107
G0 F15000 X48.427 Y10.403 Z0.35
;TYPE:SKIRT
G1 F2700 E1
```

GregValiant, 2023-05-26:
Your snippet doesn't go back far enough.
For example, if the previous XYZ location was X0 Y0 Z0.35 and it is moving to X48.427 Y10.403 Z0.35 then there would be no Z movement (Z speed = 0mm/sec) and the XY would be at a travel speed of 250mm/sec.  It is also subject to the acceleration values of the lower of either:  the Max values in the M201 firmware register, or whatever was set in Cura.  You can only go so fast in a short distance given the distances required to accelerate and to decelerate.
As @Ghostkeeper explained; the printer calculates the straight line movement from a point to a point and all the motors arrive at the location simultaneously but are constrained by M201 Max Accel and M203 Max Feedrate in the firmware.

Let's assume that your firmware Max Feedrates in M203 are X250 Y250 Z30...
Ignoring acceleration for a moment - to move 250mm from X0 Z0 to X249.8 Z10 at 250mm/sec results in a 1 second travel time. The Z moves 10mm in that 1 second so the Z speed is 10mm/sec.
To move 250mm from X0 Z0 to X176.77 Z176.77 at 250mm/sec would appear to also result in a 1 second travel time.  That won't happen because the M203 Z30 now comes into play as the speed limit and the movement would take 5.89 seconds.  Because the movement (picked with malice aforethought) is at 45°, the X speed is also 30mm/sec and not the 250 that the gcode asked for.

So the "F" value on any gcode line is constrained by the firmware settings.

Jarren-Essentium, 2023-05-27:
@GregValiant I totally agree that Marlin and Klipper and most other printer firmwares follow that approach. I don't see this as a large bug that is affecting everyone but it still is a bug in that Cura is not limiting the commanded feed rates that are part of the printer definitions. You and Cura are both assuming the printer is starting at 0,0,0 but that's not the case. The printer may be starting at Z max for it's home position of 0,0,300. So this large commanded feed rate could cause issues. What I'm trying to show is that the "theoretical bug" that ghostkeeper pointed out is still present.  My printer is not running Marlin but Marlin is the best approximation for input gcode.

Here is the full starting gcode for reference 

```
;FLAVOR:Marlin
;TIME:971
;Filament used: 1.93198m, 0m
;Layer height: 0.35
;MINX:48.145
;MINY:3.159
;MINZ:0.35
;MAXX:87.247
;MAXY:42.27
;MAXZ:19.95
;LINEARADVANCEPROCESSED
;Generated with Cura_SteamEngine 5.3.0
T0
M82 ;absolute extrusion mode
;;;;;;;;;;;;;;
;Custom Starting Script
G90
M83
M140 S80; Set Bed Temperature
M141 S0; Set IR1-Bed
M142 S0; Set IR2-Wall
T0 ; Head 0
G92 E0
T1 ; Head 1
G92 E0
M190 S80; Set Bed Temperature
M109 S300 T0 ; Set T0 hozzle for purge
M104 S100 T1 ; Set T1 hozzle for idle
T0
;;;;;;;;;;;;;;
M141 S0
M82 ;absolute extrusion mode
;;;;;;;;;;;;;;
;Tool 0 Custom Start Script
T0
M83 ; Ext Rel
M900 K0.039; K comp
M106 S100 ; Purging Fan Speed
M109 S300 T0; T0 to Temp
M420 L0 ; Topo comp enable for new head
G1 E5 F300 ; Purge T0
M104 S300 T0; T0 to Temp
M106 S30 ; Fan On
M83 ; Ext Rel
;;;;;;;;;;;;;;
M83 ;relative extrusion mode
M83 ;relative extrusion mode
G1 F2700 E-1
;LAYER_COUNT:57
M900 K0.039000 T0 ;added by LinearAdvanceSettingPlugin
;LAYER:0
M107
**G0 F15000 X48.427 Y10.403 Z0.35**
;TYPE:SKIRT
G1 F2700 E1
G1 F1800 X48.836 Y9.148 E0.08851
G1 X49.396 Y7.982 E0.08673
G1 X49.862 Y7.245 E0.05847
```

GregValiant, 2023-05-27:
What I'm saying is that it doesn't matter what the F rate is in the line.   If the limits of M203 are exceeded then the firmware will control the speed of the move
If the M203 limits of any axis are not exceeded by the required speed of the move then the move will indeed be at F15000.

You are right in that there is no way to tell where the printhead is moving from at the end of leveling, just that it is going to X48.427 Y10.403 Z0.35.

This code assumes your print volume is at least X200 wide and Z205 high.  you can paste it into a text file and save it as a gcode and print it.  I think it adequately shows what will happen in two different moves at a requested 250mm/sec

G28 ;HOME
M203 Z10 ;set max Z feedrate for this demonstration
G0 F600 Z5 ;MOVE UP OFF THE BED
G0 F15000 X200 Z15 ;move at 250mm/sec
G0 X0 Z5 ;move back at 250mm/sec
G0 F15000 X200 Z205 ;move up at Max Z per M203
M203 Z30 ;reset Max Z speed
G0 F15000 X0 Z5 ;move back at Max Z speed
M84

When you turn your printer off and then back on it will reset M203 back to it's defaults.
The printer definition files contain default information to set Cura up for a printer.  Those settings fill in boxes in Cura but have no effect on what the actual printer settings might be and there is no way for Cura to know what those settings might be.

Jarren-Essentium, 2023-05-27:
@GregValiant believe me when I say I really appreciate all your work on Cura and these forms. I just want to let it be documented that this is a bug and something to look at in the future. My firmware doesn't support M203. And this line of Gcode causes axis errors on this printer. I ran your gcode and it failed the motion command with many errors. My printer is closer to a CNC machine. I have found a work around to make these high Z feedrates work with specific find and replace for Z commanded motion above the feedrate but the fact is still that the Cura engine is commanding feedrate above what is specified in the definition files.

GregValiant, 2023-05-27:
Alright, now I have new information.
It sounds like you have a rather clumsy workaround for the fact that M203 is not supported by your firmware.
Understand that I don't work for Ultimaker and I don't have an UltiMaker printer.  I know that UM will not pay their people to fix Cura so it is perfect for my Creality printer.  They also will not go out and buy one of every printer on the planet and adjust Cura to fit some other manufacturer's printer.  Neither of those make any sense business wise.  I'll quote one of the Cura developers who said in a post "There's only so much we're willing to give away for free.".  That's something I can understand.

If your firmware absolutely has no equivalent of M203 then I think your best bet would be a dedicated Post Processor that would go through the Gcode and calculate the length of every move, and adjust the speed when needed so that no axis will exceed limits you set.  It would effectively become your "On-the-Fly Speed Limit" as M203 is in most dedicated FDM firmwares.

You have a third party printer with what sounds like non-mainstream firmware.  If you want to consider it a bug that's fine.  I just don't see it being adjusted by UM as they would have to pay their people to change it.

MariMakes, 2023-06-01:
Hey @Jarren141,

UM team here 👋. 
I discussed your issue with the team and we discussed some possible workarounds to get you unstuck but we came to the same conclusion with post-processors. 

@GregValiant is correct. Since it's an edge case only on a printer that we don't actively maintain, so we cannot resolve it on the boss's time. Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I'll close this issue. I hope you understand 😇

closed 2023-06-01 by MariMakes

Jarren-Essentium, 2023-06-01:
@MariMakes I understand totally. Thanks to the whole team for the work that you all do.