Ultimaker/Cura #10233: Add option to always enable z-hop for combining if travel is higher than max comb distance with no retract

Pla5ma, 2021-08-05:
### Is your feature request related to a problem?

Yes. If a print is having closed inner walls the print head needs to cross the top/bottom layers in any case and this creates scratches on those surfaces.

### Describe the solution you'd like

Add an option to always enable z-hop for combining if travel is higher than max comb distance with no retract.

### Describe alternatives you've considered

None.

### Affected users and/or printers

All and its a topic which pops up in many Cura discussion groups. Many people are disabling combining due to this.

fvrmr, 2021-08-06:
Hi @Pla5ma thank you for your feature request. 
I will discuss this with the team. Keep you posted

fvrmr, 2021-08-12:
I have created a ticket on our backlog for this feature request. 
Devs see CURA-8478