Ultimaker/Cura #10234: Please don't auto-generate "connective" supports

andersontba, 2021-08-05:
### Is your feature request related to a problem?

I'm frustrated when Cura generates unnecessary supports that make removal especially difficult -- specifically connecting support material in spaces where no supports are needed, linking regions that I'd like to be able to break out separately.  With tougher materials like PETG, breaking up these supports that are locked inside is even more difficult.

### Describe the solution you'd like

The attached image is of a GoPro tripod mount.  While the three horizontal holes need support, the empty regions between the three vertical flanges do NOT need support -- but by building these tall blue supports between the flanges, the support material gets locked in there, and I can't remove it easily.  I'd like the support to JUST be created in the holes, as independent small pieces that can be broken out more easily.
[image]

### Describe alternatives you've considered

Extensive cutting, grinding, etc. to remove these is a painful workaround for such an obvious problem.  I see something into here about "support blockers", and I'll go mess around with that now, but Cura shouldn't be generating these supports in the first place.  There's just no reason for them.

### Affected users and/or printers

Anyone doing relatively intricate parts with multiple overhang regions in close enough proximity that Cura is connecting them today.

GregValiant, 2021-08-05:
There are a lot of Support Settings along with the Support Blockers.  What you are getting just needs to be tuned.  There is no "one size fits all" so you need to customize each print.  I think you'll find the tools are there.  Support Blockers can be configured numerous ways and scaled, moved, and rotated like any model.
Next to the settings Search box is a drop down for setting visibility.  Set it to "All" and go down to the Support section to see what can be done to get it the way you want.  You might want to try Tree supports with Build Plate Only so a support doesn't grow from your part.

Asterchades, 2021-08-06:
I'd be genuinely curious to see what your support settings are. I agree that there's no reason the supports should be getting generated between the structures that require the supports, since the supports themselves should only be getting generated based on what's directly above them - not what's to their sides.

You're obviously not using any form of horizontal expansion on them, nor conical supports, so there's something else causing it to stitch the pieces together and generate extra, superfluous support structures. If I were to take a guess, I'd say you're possibly looking at the **Support Join Distance** (possibly should be lower) or **Use Towers** (possibly should be disabled) settings.

fvrmr, 2021-08-06:
Hi @andersontba in experimental you could also try `break up support in chunks` under experimental. But also what @Asterchades mentioned to take a look at `support join distance`.

yet-another-average-joe, 2021-08-09:
@andersontba Sorry if I'm not really answering to the question and I'm a bit out of topic, but I designed and made many GoPro mounts. First of all you don't need any support for such a part if you print it this way. Just slightly oversize the 5mm holes (~5.5mm). You could also use a drill. The hexagon has to be oriented so top and bottom will be horizontal (they will bridge great). Also, oriented this way, chances are the mount will be fragile, even with 100% infill. It should be printed rotated 90°, the holes remaining horizontal, with supports touching buildplate, with a roof, and a two layer Z distance between roof and part. This helps a lot when removing them !

closed 2021-08-20 by no-response[bot]