Ultimaker/Cura #10235: Filament Cost

nissenhj, 2021-08-05:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Sovol-SV01

### Reproduction steps

When you enter a price under "Filament Cost", the price changes for all filaments that are in that group!
This can not be r
[image]
ight as it does not correspond to reality. 

### Actual results

The price changes for all filaments in the group
[image]

### Expected results

The price should only be changed for the filament I am in and change

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

has no more files 
[image]
[image]

fvrmr, 2021-08-06:
Hi @nissenhj you can unlink your materials if you duplicate a material from another material. 
[image]

Let me know if that works for you!

nissenhj, 2021-08-06:
> Hi @nissenhj you can unlink your materials if you duplicate a material from another material.
> [image]
> 
> Let me know if that works for you!
Hi

I have not such a button?

[image]

fvrmr, 2021-08-12:
Could you try to duplicate one and see if the button is there after you clicked duplicate?

nissenhj, 2021-08-12:
Hi
I have tried to duplicate one but no button appeared 
I am running Cura 4.10.0

fvrmr, 2021-08-13:
Can you share your material files with me? You can find them here: `C:\Users\<your username>\AppData\Roaming\cura\4.10\materials`

nissenhj, 2021-08-15:
Yes I will  share this files with you. But I don´t know how to do in here?

nissenhj, 2021-08-15:
[Cura_4_10_0_materials.zip](https://github.com/Ultimaker/Cura/files/6988746/Cura_4_10_0_materials.zip)
So! I made a zip file of it

Ghostkeeper, 2021-08-17:
The materials in question all have the same GUID so they are indeed linked and it is correct that they share their cost. They are meant to denote different profiles for the same spool of material.

However it is then incorrect that it doesn't display the button to unlink them.

It's not displaying the button due to this error, which shows up in the log:

    MaterialsView.qml:34:5: Unable to assign QStringList to QString

I've fixed that error now. The unlink button should properly show up again.

closed 2021-08-17 by Ghostkeeper

nissenhj, 2021-08-28:
Thanks for any help!

commit 33edc4a8 (2021-08-17): Fix display of unlinking button

It was not being displayed with Qt 5.15 due to the following error:
MaterialsView.qml:34:5: Unable to assign QStringList to QString
This is correct. And it should not just concatenate all of these material names; it should add a comma between them for human-readable display in the text label above the button.

Fixes #10235.