Ultimaker/Cura #10236: estimated time doubled all of a sudden

jledcustoms, 2021-08-06:
### Application Version

4.9.1

### Platform

MacOS 10.14.6

### Printer

Ender 5 plus

### Reproduction steps

All of a sudden Cura is telling that the estimated time of a print is double. 
I've printed this part numerous times and it has always been 21hrs to complete. 
I use a preset when I import the .stl and slice
I made a small change to the .stl and after that the estimated print time went to 1 day 18hrs
I closed Cura and testing on multiple computers with the same result
I then opened the back up (original)  .stl and that same 1 day 18hrs.
I've tried multiple versions of Cura and the same. I've gone into settings and ticked ALL settings viewable and gone through ALL of them and cannot figure this out. 

Cura did show me the "what's new" pop up at one point leading me to thing there was an automatic update? version still stated 4.9.1 though

I have also downloaded Prusaslicer and confirmed with the same settings the print time is around 20hrs. 

I'm loosing my mind as to why this is happening. and support will be gladly accepted otherwise I will have to transition to PrusaSlicer

### Actual results

print time is still 1 day 18hrs

### Expected results

21hrs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional information

jledcustoms, 2021-08-06:
Also it takes a good 10minutes to slice this file

fvrmr, 2021-08-06:
Hi @jledcustoms could you share a project file (file - save project) then I can take a look at it.

jledcustoms, 2021-08-06:
> Hi @jledcustoms could you share a project file (file - save project) then I can take a look at it.

Is this a private forum? Or is this open to the public? I cannot share the file publicly

fvrmr, 2021-08-06:
@jledcustoms no this is not a private forum, but I saw you posted as well on our 3D printing forum. I will contact you there.

Ghostkeeper, 2021-08-17:
@fvrmr Did you get in contact with Jledcustoms about the model used in [their forum thread](https://community.ultimaker.com/topic/37694-estimated-time-doubled-with-no-changes-to-cura-or-stl/)?

Is there something we can do to resolve their issue?

fvrmr, 2021-08-17:
@Ghostkeeper yes I am in contact with @jledcustoms 

It is really difficult to see what is causing this, since I don't have anything to compare with. On a Ultimaker S5 it also slices with a time estimation of 1 day and 7 hours.

closed 2024-12-09 by github-actions[bot]